RIOT-OS/RIOT #5867: telosb: radio initialization fails

yschroeder, 2016-09-26:
When using a TelosB (in this case actually a TMote Sky) with the `default` example the radio does not work.

Enabling debug messages in the cc2420 driver says `cc2420: init: unable to communicate with device`. The TelosB board never initializes `CC2420_PARAMS_BOARD` and uses the default parameters from the driver (which obviously fails).

Side question: Why does the CC2420 driver (the AT86RF2xx does this aswell) have default pin configurations? What is the use case here? Shouldn't it just throw an error, when a driver is loaded but the pins are undefined? I think having those defaults is rather dangerous as it may reconfigure pins used otherwise and can cause hardware failure.

PeterKietzmann, 2016-09-27:
A default configuration for the cc2420 on the telosb node is indeed a missing feature as I see it. And I assume (as you obviously do as well) that the drivers default config is not correct for the telosb. Are you familiar with that device and willing to submit a configuration :-)? Otherwise @kYc0o would you provide one? 
I'm not 100% sure but as far as I know the drivers-default configuration was just to have a pre-defined setup for externally connected devices.

yschroeder, 2016-09-29:
I already tried to make a default configuration, but it did not work straight away.
Here is what I currently got:

```c
#define CC2420_PARAMS_BOARD   {.spi        = SPI_0, \
	                           .spi_clk    = SPI_PIN_CLK, \
	                           .pin_cs     = GPIO_PIN(P4, 2), \
	                           .pin_fifo   = GPIO_PIN(P1, 3), \
	                           .pin_fifop  = GPIO_PIN(P1, 0), \
	                           .pin_cca    = GPIO_PIN(P1, 4), \
	                           .pin_sfd    = GPIO_PIN(P4, 1), \
	                           .pin_vrefen = GPIO_PIN(P4, 5), \
	                           .pin_reset  = GPIO_PIN(P4, 6)}
```

However, this does not seem to work (or something else is broken in my setup...).
I used [this datasheet](http://www4.ncsu.edu/~kkolla/CSC714/datasheet.pdf) as reference.

Regarding the default configurations: Should I open another issue to have a better place for discussion?

kYc0o, 2016-09-29:
I'm looking at the code and will try to get the hardware for testing, although it seems complicated...

PeterKietzmann, 2016-10-05:
@kYc0o is the code okay?

kYc0o, 2016-10-05:
The code seems OK and the pin configuration too... I'd like to have the hardware to make the tests locally, but unfortunately I don't have it... @haukepetersen maybe has one?

thomaseichinger, 2016-10-05:
This message indicates that SPI communication is not properly set up. Can you add a `DEBUG("%x\n", reg)` and see if the read data is anything else than `0xffff` or `0x0000`?

PeterKietzmann, 2016-10-06:
Yes SPI (I/O) is not working correctly. I'm will look into it (unfortunately) and will report back once I found something.

PeterKietzmann, 2016-10-07:
Would be great if you could test #5917 with #5918

yschroeder, 2016-10-12:
I tested both PRs together on two TMote Skys. I used the default example and transmitted data between both of them via `txtsnd`. Sending and receiving both work. The PRs fix this issue!

:+1:

PeterKietzmann, 2016-10-12:
Great to hear. Thanks for testing

haukepetersen, 2016-10-17:
#5918 and #5917 were merged, so is this issue fixed now in master?

PeterKietzmann, 2016-10-17:
yes

closed 2016-10-17 by PeterKietzmann