RIOT-OS/RIOT #5888: how to set netif parameter

immesys, 2016-09-28:
Say I want to set a parameter on the radio device, specifically something like

```
netopt_state_t sleepstate = NETOPT_STATE_SLEEP;
driver->set(dev, NETOPT_STATE, &sleepstate, sizeof(netopt_state_t));
```

but I would like to do this from the application where there is no handle to the driver, and I want it to go through the queues so that if I do wakeup->send packet->sleep that the order happens correctly. How do i  do this? I understand this is the role of a MAC layer but at this point I would like to control it from the app.

Some things I have tried:
send a message to the radio driver: the driver handle is static inside auto_init so I can't get the PID
send a message via `gnrc_netapi_dispatch_send` with GNRC_NETTYPE_NETIF : not clear what the pktsnip would be, or if this is legal.

I'm sure this is simple for someone familiar with this part of the code, like @kaspar030 or @haukepetersen

miri64, 2016-09-29:
It seems you are using GNRC, so why not use `gnrc_netapi_get()` with the interface the device is using? The number the interface is identified by is its PID. `GNRC_NETTYPE_NETIF` can't be registered to `gnrc_netreg` because in the usual stack use-cases there is no need for that (interfaces are not "receiving" as they are directly communicatingwith the driver and for sending they are identified by the FIB anyway),  so we saved the memory for those.

miri64, 2016-09-29:
> GNRC_NETTYPE_NETIF can't be registered to gnrc_netreg

This means you can't use `gnrc_netapi_dispatch_send()` (since this looks up the PID in `gnrc_netreg`), but nobody stops you from just using `gnrc_netapi_send()` but this has nothing to do with options, so I'm not sure how this is related to the question.

miri64, 2016-09-29:
You can get all interfaces available using [`gnrc_netif_get()`](http://doc.riot-os.org/group__net__gnrc__netif.html#gae81e49c4abc7735987200d880ddab2fd). [Here](http://doc.riot-os.org/group__net__gnrc__netapi.html#ga0ca19b514ad9303c0c3bc206fbe15b0a) is a link for `gnrc_netapi_get()`, too ;-).

immesys, 2016-09-29:
OK, thanks for the pointers! That ought to work. Sorry for the noise, it can be difficult to work out the best way to implement things, and now that I'm trying to upstream all my work I find it pays to ask first so that the code ends up fitting in properly later.

closed 2016-09-29 by immesys