RIOT-OS/RIOT #5889: doc: @defgroup's name should reflect the modules name

miri64, 2016-09-29:
Looking just at the [generated documentation](http://doc.riot-os.org) I myself find it very hard to find a module. Some examples include:

* `gnrc_netapi` => "Generic (GNRC) network stack => GNRC communication interface"
* `xtimer` => "Timer"
* and many more

I propose to require the `@defgroup` command as follows: `@defgroup <module_name> Module name` with <module_name> being the canonical name we decided in https://github.com/RIOT-OS/RIOT/issues/2422 on, and "Module name" being a human-readable version of the same canonical name. If it is a submodule the prefix can obviously left out of the "Module name".

OlegHahm, 2016-09-29:
Hm, not sure I understand the problem. If someone is new to RIOT and wants to use a timer, he/she's probably looking for "Timer". Hence, it's helpful (although not 100% required) if the documentation about the timer module can be found at the alphabetical position of "Timer". If someone like you or me, who knows RIOT quite well is looking for something like netapi, I can simply use the search function of the doxygen webinterface for netapi or go directly to the file view. I think mixing code and documentation names, would look strange to me.

miri64, 2016-09-29:
Mh, from that perspective this might make sense...

closed 2016-10-17 by miri64