RIOT-OS/RIOT #5896: native board segfaults when module xtimer is loaded with glibc < 2.17

yschroeder, 2016-09-30:
xtimer uses a call to `real_clock_gettime()` in its depths which should map to `clock_gettime()` from `time.h`. However, on machines with glibc < 2.17 `dlsym()` for this function silently fails (not even giving an error in `dlerror()`) and returns a null pointer for that function. This results in a segfault during initialization of the xtimer module.

Linking with `-lrt` fixes this problem and `real_clock_gettime()` will point to the correct function. The manpage for `clock_gettime()` also states that `-lrt` is needed for glibc < 2.17.

The build process for native (or xtimer...) should check the libc version and link librt if needed.

I found this problem the hard way while debugging on a Raspberry Pi with Wheezy (libc 2.13).

OlegHahm, 2016-09-30:
I tend to say: wontfix and add the 2.16 to the [not supported toolchains](https://github.com/RIOT-OS/RIOT/wiki/Family%3A-native#working) for native.

yschroeder, 2016-10-04:
Yes, for me this is sufficient to solve the issue.

However, I wanted to report it so maybe other people running into the problem can find this issue and solve their problem faster.

miri64, 2016-10-18:
Done

closed 2016-10-18 by miri64

yschroeder, 2016-10-19:
I just learned that the native board's makefile covers this problem [here](https://github.com/RIOT-OS/RIOT/blob/master/boards/native/Makefile.include#L126).

The only thing is that this probably belongs not into the board but into the cpu. I think in my case it broke because I made a new cpu based on native with another name and that broke it because the Makefile explicitly checks if the cpu's name is native.