RIOT-OS/RIOT #5919: Is there any examples for the RPL over nodic 51822 BLE?

desword, 2016-10-09:
I have noticed the implementation of RPL of RIOT. However, is there any example application for nodic 51822 BLE? Thanks.

miri64, 2016-10-09:
Hi, the 6Lo implementation of RIOT was also written with [RFC 7668](https://tools.ietf.org/html/rfc7668) in mind, but since at the point of implementation there was no BLE support in RIOT yet, it might not work since nobody (as far as I know) tested it.

miri64, 2016-10-09:
(If 6Lo works, IPv6 works, so RPL should work, too ;-))

miri64, 2016-10-18:
Maybe @kaspar030 can answer the question from the BLE perspective?

OlegHahm, 2017-01-15:
I think the answer is: probably not out of the box, but please feel free to test. Concrete issues can be discussed in a separate discussion here or on the mailing list.

closed 2017-01-15 by OlegHahm