RIOT-OS/RIOT #5928: xtimer: usage in board_init() crashes

basilfx, 2016-10-11:
Initializing and using xtimer before the kernel has initializes leads to a crash (no output). I verified this for the Thunderboard Sense (EFM32) target and Seeeduino Arch Pro (LPC1768). I use it exactly the same as the Mulle board (#4133).

The problem does not happen when the delay is less than the backoff. So my workaround for sleeping 1ms is sleeping 100 times 10 µs.

Use case: The Thunderboard Sense has a I2C GPIO expander that needs 1ms wakeup time. I address this GPIO expander to enable sensors so they get initialized before `auto_init()`.

miri64, 2016-10-19:
Why not use `periph/timer` for that?

basilfx, 2016-10-20:
I don't have timers available, because the only timer available is reserved for xtimer.

Furthermore, it would be a workaround. Take #4133 as an example: if it would depend on `periph/timer`, it would make the nvram driver not less portable.

kaspar030, 2016-10-20:
xtimer as is expects threading to be initialized, and that can only be changed with unacceptable runtime overhead.

> The problem does not happen when the delay is less than the backoff. So my workaround for sleeping 1ms is sleeping 100 times 10 µs.

That's because with these short delays, xtimer spins instead of issuing a task switch.

Why not work around the issue?

At startup time, using a simple ```volatile unsigned N = <number of ticks>; while(N--)``` for 1ms is IMHO perfectly acceptable. Or, move the expander initalization into auto_init() after xtimer.

basilfx, 2016-10-20:
I'll go with the while-loop, as auto_init() would add board-specifics to the RIOT-OS's system.

But then again, #4133 is wrong (cc @gebart). If one would change the timings in the nvram driver, the mulle board would also crash.

kYc0o, 2017-04-25:
Can we close this issue then?

basilfx, 2017-10-24:
Yes :-)

closed 2017-10-24 by basilfx