RIOT-OS/RIOT #5930: native: `ncache add` shell command results in segfault

haukepetersen, 2016-10-12:
Found a way to cause a segfault in 'native': run the default example and simply call `ncache add` without any further argument -> segfault:
```
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: )
RIOT network stack example application
All up, running the shell now
> ncache add
ncache add
make: *** [term] Segmentation fault
```

I don't have time to look into this right now, but maybe we should fix this?! :-)

closed 2016-10-12 by miri64

LudwigKnuepfer, 2016-10-16:
@miri64  So this was not an issue with native but with the shell, right?

miri64, 2016-10-16:
Yapp, @haukepetersen just encountered it on native.

PR #5931 (2016-10-12): shell: ncache: Fix #5930
As it says in the title: fixes #5930
commit 4e8659c7 (2016-10-12): shell: ncache: Fix #5930
commit 5d9e7662 (2016-10-12): Merge pull request #5931 from miri64/sc_ipv6_nc/fix/5930

shell: ncache: Fix #5930