RIOT-OS/RIOT #5932: sock: timeout/non-blocking behavior for send

miri64, 2016-10-12:
With #5929 basically non-blocking communication was introduced to `sock`, at least for received. But this kept me thinking: Should we introduce that feature for `send`/`write`, too? Up until now we only have blocking `send`, but no timeout, no non-blocking capabilities.

miri64, 2016-10-18:
@kaspar030 ping? **Now** is the is the time for easy changing ;-)

kaspar030, 2016-11-18:
Hm, didn't we add the flags field on create for non-blocking operation?

miri64, 2016-11-18:
That would be a different paradigm than we have for receive...

kaspar030, 2016-11-18:
It is also solving a different problem.
For UDP, the receive timeout specifies "how long to wait for a packet sent by the remote". For send, it means "how long to wait at most until there's free space in the send buffer". In the latter case, it is perfectly valid to just drop the packet.

For TCP it's a different story, as for sending the congestion control plays a role. POSIX has SO_RCVTIMEO/SO_SNDTIMEO as socket options...

closed 2022-04-18 by stale[bot]

miri64, 2022-04-18:
Maybe of interest to @maribu or @benpicco still?

maribu, 2022-04-19:
Note that currently we practically have non-blocking send with GNRC, until `gnrc_tx_sync` is implemented throughout the stack. (I am fully aware that I am stalling my PRs right now that would provide these pieces. I shifted them up on my priority list once more and hope that this is sufficient to allocate some time for it soonish.)

But I bet some use cases will break that rely on the current (semi-) non-blocking behavior when one starts using `gnrc_tx_sync`. So I think we should provide a way to decide at runtime if one needs non-blocking or blocking behavior, so that code relying on different ends of that scale can coexit.

benpicco, 2022-04-19:
Something like `O_NONBLOCK` would probably be useful on that case.

maribu, 2022-04-19:
But I agree with @miri64 that adding an argument to the send function - or maybe providing a second function - would be more convenient to use than `setsockopt()` and be a more consistent API. The `setsockopt()` style adds a somewhat "hidden" state, so that to identical calls to send in identical situations behave vastly different based on that state. To me, this is non-obvious and easily allows some function to modify that state so that code not prepared for non-blocking behavior is executed on a non-blocking socket. An explicit argument (or function name) makes the expected behavior much more obvious and the code is easier to review, as the reviewer no longer has to scroll through the code to check whether or not the socket is blocking or non-blocking.

Note that if we would provide two functions for blocking and non-blocking behavior, those could internally be implemented as two `static inline` wrappers for a function with an additional flags parameter.