RIOT-OS/RIOT #5934: drivers: at86rf23x address endianness

immesys, 2016-10-12:
When setting the long address from the cpuid, the existing code switches the byte order [(line 90)](https://github.com/RIOT-OS/RIOT/blob/master/drivers/at86rf2xx/at86rf2xx.c#L90)

I am not convinced this is correct. A few lines above, we manipulate the first byte to set the multicast /global unique bits but after the byte order switch, that byte is now the least significant byte, so either we manipulated the wrong bits or we should not be switching the byte order. My feeling is the latter, because while doing a port for a platform that has a real EUI64 as the CPUID, I discovered that the generated address had the wrong byte order (and did not have the correct bits flipped for global unique / multicast). 

Am I missing something? Or should I PR to remove the NTOH in at86rf2xx_set_addr_long ?

thomaseichinger, 2016-10-12:
I am always getting a bit dizzy with all the endianness changes but I think the existing code, in sum, is correct. The thing is, `addr_long.uint64.u64` is stored in network byte order (`network_uint64_t`) so we need to use `NTOH` to avoid compiler errors. The confusing part is the loop in `at86rf2xx_set_addr_long()` where the address once gets assigned to the internal struct `dev->netdev.addr_long` in host byte order and bit shifted `(addr >> ((7 - i) * 8))` to be set on the radio module. `(AT86RF2XX_REG__IEEE_ADDR_0 + i)` 

Hope that helps (and I actually got it right :grin:)

immesys, 2016-10-12:
Ok, say for example the CPUID is 8 bytes long, and is `11 12 13 14 15 16 17 18` with `11` being the least significant byte. This code will then change the `11` to `12` (clear bit 0x01, set bit 0x02), and then flip the bytes in the NTOHLL. The resulting LL address would be `fe80::1817:1615:1413:1212` when it should actually be  `fe80::1a17:1615:1413:1211` according to appendix A of RFC2373. 

So I think, if we are both right, then I need to change my CPUID byte order so that it is correct after the NTOHLL, but we should change the code to manipulate cpuid[7] instead of cpuid[0] so that after the flip the right bits are fiddled.

immesys, 2016-10-19:
@thomaseichinger I know things are a little busy at the moment, but I don't want to forget about it. Does my example make sense?

thomaseichinger, 2016-10-21:
@immesys I'm very sorry for the delay, I will look into it in the course of the day.

immesys, 2016-11-21:
I'm back from conferences and whatnot. @thomaseichinger do you want to ping me offline when you have a moment to discuss this issue?

OlegHahm, 2016-12-08:
@thomaseichinger, do you?

closed 2019-09-10 by stale[bot]