RIOT-OS/RIOT #5959: byteorder functions in capital letters?

kaspar030, 2016-10-17:
RIOT's ```byteorder.h``` defines ```NTOHL()``` as capital static inline functions. Does anyone remember the reason for not using ```ntohl()``` et al?

Like this it is really inconvenient for code also used on other platforms (see https://github.com/kaspar030/sock/issues/2).

miri64, 2016-10-18:
@kaspar030 I guess because our coding conventions state that macros should be upper case. For portability you can find the POSIX naming in `sys/posix/include/sys/bytes.h` ;-)

miri64, 2016-10-18:
(which also should help you with portability since the macro is exactly where other OSs would expect it to be `<sys/bytes.h>`)

kaspar030, 2016-10-18:
> I guess because our coding conventions state that macros should be upper case.

Well, that's the thing. They are static inline functions, not macros.

miri64, 2016-10-18:
See https://github.com/RIOT-OS/RIOT/pull/4812

kaspar030, 2016-10-18:
So with the current state of byteorder.h, you don't remember the reason for the function names being uppercase?

miri64, 2016-10-18:
Not really, but I guess the HTONS etc. wrapper macros are primarily there for legacy reasons. @Kijewski?

kaspar030, 2016-10-18:
> the HTONS etc. wrapper macros

They're not macros. :)

miri64, 2016-10-18:
That's even more confusing ^^.

gebart, 2016-10-18:
I think they were macros at first but were changed into inline functions to
make it easier to see programming mistakes.

kaspar030, 2016-10-18:
I would propose then, independently of #4812, to change them to use lowercase names. #4812 can then implement the functions using the const macros (or builtins...), and the posix wrapper can just include byteorder.h. What do you think?

miri64, 2016-10-18:
Fine by me.

OlegHahm, 2017-01-15:
I would suspected that non-RIOT code rather expects these functions to be macros and thus to be uppercase.

kaspar030, 2017-04-13:
> I would suspected that non-RIOT code rather expects these functions to be macros and thus to be uppercase.

Just checked, in my system headers they are declared as functions.

OlegHahm, 2017-04-13:
True.

miri64, 2017-04-13:
According to the [POSIX-specification](http://pubs.opengroup.org/onlinepubs/9699919799/functions/htons.html) they are functions.

PR #6897 (2017-04-13): core: byteorder.h: rename NTOH* -> ntoh*, HTON* -> hton*
Fixes #5959.
closed 2017-04-13 by miri64