RIOT-OS/RIOT #5960: objcopy: command not found

sinkarharshad, 2016-10-17:
Hi,

I'm trying to setup RIOT shell on MAC OSX using this guide: http://riot-os.org/api/index.html#the-quickest-start

While trying 'make all' I'm getting an error:

__TEXT	__DATA	__OBJC	others	dec	hex
90112	1220608	0	24576	1335296	146000	
bash: objcopy: command not found
make: *** [all] Error 127

And there's an error after using the command: 'git checkout <LATEST_RELEASE>'
Error: -bash: syntax error near unexpected token `newline'

I installed binutils but the package name is gobjcopy rather than objcopy. I'm unable to use it. 
I also added alias in ~/.profile as:

alias objcopy="gobjcopy"
alias objdump="gobjdump"

This also didn't work and I got the same error that objdump command is not found.

Please let me know how to proceed from here

Yonezawa-T2, 2016-10-18:
If you are using MacPorts, setting `PATH` will work.
`PATH=$PATH:/opt/local/x86_64-apple-darwin15.6.0/bin/ make BOARD=native`

sinkarharshad, 2016-10-18:
I tried, and still I get the same error. ): 
[image]

Yonezawa-T2, 2016-10-18:
Type `mdfind -name objcopy` to locate where the `objcopy` is installed.

If you don't have a file named `objcopy` but `gobjcopy`, make a symlink to it and add it to your `PATH`:
```
mkdir -p ~/bin
ln -s /path/to/gobjcopy ~/bin/objcopy
ln -s /path/to/gobjdump ~/bin/objdump
...
 PATH=$PATH:~/bin/ make BOARD=native
```

sinkarharshad, 2016-10-18:
I have installed objcopy. what should i do next? (Below is the path)
[image]

Yonezawa-T2, 2016-10-18:
```
mkdir -p ~/bin
ln -s /usr/local/Cellar/binutils/2.27/bin/gobjcopy ~/bin/objcopy
PATH=$PATH:~/bin make BOARD=native
```

Yonezawa-T2, 2016-10-18:
To other maintainers:
BTW, `boards/native/Makefile.include` will use `gobjcopy` if it is available, but `cpu/Makefile.include.llvm` just try to use `objcopy`. I guess that is the cause of this issue (not confimed yet). If so, we should update `cpu/Makefile.include.llvm` to use `gobjcopy` if `objcopy` is not available.

sinkarharshad, 2016-10-18:
After following  above instructions, again when i tried running "make all" i still got the same error.  :! (or after following above instruction, I don't need to run "make all" ? )

kYc0o, 2016-10-18:
I confirm the issue, I didn't provide a patch for that since I don't use a lot the native port but I think we should work on it, even if there are no "a lot" of Mac OS X users I think is important.

miri64, 2016-10-18:
@kYc0o would it be possible to provide this patch for the release?

Yonezawa-T2, 2016-10-19:
@sinkarharshad 
What is displayed with the following commands?

```bash
ls -l ~/bin
export PATH=$PATH:~/bin
type objcopy
```

Yonezawa-T2, 2016-10-19:
@miri64 @kYc0o I made a PR #5965.

sinkarharshad, 2016-10-19:
@Yonezawa-T2 "make all" no error, for "make term" I am getting this following error. 
[image]

Yonezawa-T2, 2016-10-19:
@sinkarharshad Install http://tuntaposx.sourceforge.net/.

sinkarharshad, 2016-10-19:
After installing tunetap, and I tried again .. and got following error. 
[image]

Yonezawa-T2, 2016-10-19:
Try `sudo make term`.

Tips: if you get `resource busy` error on tap interface, find a process named *.elf, then kill it.

sinkarharshad, 2016-10-19:
It worked, and finally got a message saying .. "Welcome to RIOT" .. thank you all for religiously putting your efforts helping me. Thank you one more time. Cheers! Also, As I am beginner to riot, can you please guide me? How should i start writing simple programs on RIOT? Thank you!

miri64, 2016-10-29:
https://github.com/RIOT-OS/RIOT/pull/5965 is not merged yet

closed 2016-10-31 by miri64

diskmila, 2022-05-02:
Any updates on this matter? I'm having the same problem

kYc0o, 2022-05-03:
So far it was fixed on #5965, what's your exact problem?

diskmila, 2022-05-04:
when running make all and make term from the tutorial I get the objcopy not found error
[image]

PR #5965 (2016-10-31): make: use gobjcopy if objcopy is not available
Fixes #5960

BTW, `make TOOLCHAIN=llvm` results in `error: unknown target triple '-Wall', please use -triple or -arch`. We should investigate this later as another issue.