RIOT-OS/RIOT #5962: Makefile: UNDEF variable is not working as documented

yschroeder, 2016-10-18:
As I understand the documentation for `UNDEF` in [Makefile.vars](https://github.com/RIOT-OS/RIOT/blob/master/Makefile.vars#L44) (which makes not too much sense), the `UNDEF` variable should undefine object files to avoid their use in the final executable to avoid multiple definitions of functions.

In my case I want to override the gpio implementation of the native cpu with my own implementation in the board.

However, the `UNDEF` variable works the other way around. It is given as an extra parameter to the final gcc call in [Makefile.include](https://github.com/RIOT-OS/RIOT/blob/master/Makefile.include#L267). This results in the explicit use of the undefined files by the linker!

I think `UNDEF` should be used somewhere around [here](https://github.com/RIOT-OS/RIOT/blob/master/Makefile.base#L44). It should filter the objects.

kaspar030, 2016-10-18:
Let's get rid of it, see https://github.com/RIOT-OS/RIOT/pull/5757.

yschroeder, 2016-10-18:
Well, if that functionality is removed, how could I solve my use case?

A bit more info on it:
I have a board that uses the native cpu but should provide real gpio functionality. The native cpu has a dummy implementation for gpio that does nothing. I now want my board's implementation to be used instead of the one in the native cpu. If this a bad idea altogether then please point me into a direction that makes more sense...

kaspar030, 2016-11-18:
> I have a board that uses the native cpu but should provide real gpio functionality. The native cpu has a dummy implementation for gpio that does nothing. I now want my board's implementation to be used instead of the one in the native cpu.

Are you using Linux's interfaces for accessing the gpios? If yes, that could become the default gpio implementation.

yschroeder, 2016-11-21:
Currently I do not do that. I'm on a Raspberry Pi and found it easier to use specific libraries. Standard GPIO functions should not be that complicated with the Linux interfaces but I don't know yet if it's possible to have (good) interrupts and callbacks for externals hardware with that. I am currently using PIGPIO (http://abyz.co.uk/rpi/pigpio/). That library has an interface that is pretty compatible to the one from RIOT (interrupt callbacks with extra arguments are possible directly).

cladmi, 2018-01-26:
Added a PR to fix the documentation https://github.com/RIOT-OS/RIOT/pull/8460.

Which does not solve your problem, but matches the issue title.

kYc0o, 2018-07-17:
> Which does not solve your problem

So what would do it? @yschroeder is still a problem for you with the newly added documentation?

yschroeder, 2018-07-19:
Oh, this a very old one... I'm not working on my Raspberry Pi RIOT port anymore, so for me this issue is obsolete. Don't know if this would still be a problem for other people though.

I guess it would be a good idea to add a Raspberry Pi with native CPU and allow the use of the GPIOs pins.

cladmi, 2018-07-25:
I think the original problem would not be "replace an object from the compilation list" but more "use another gpio MODULE" implementation.
I do not think I would add support for doing changes inside of a module in a "hacky" way, but taking into account that some module should be selectable, even with `periph-` is important.

I will currently close this issue as this original described issue is addressed. Please re-open a dedicated issue if needed.

I think @bergzand had something for the raspberry-pi gpios

closed 2018-07-25 by cladmi