RIOT-OS/RIOT #5967: saml21-xpro: i2c compilation error

dylad, 2016-10-19:
Hi,
I see that some code has been shared between saml21 & samd21 MCU.
I tried to build an application with I2C on saml21-xpro but I got several errors because CMSIS files of saml21 have not been added to RIOT yet. 
I take a look to ASF and find the required files.
There are some difference between cmsis/samd21/include/component/sercom.h & cmsis/saml21/include/component/sercom.h so it seems to be not possible to use directly the sercom.h of saml21 with the current i2c.c file in sam21_common folder.
regards

PeterKietzmann, 2016-10-19:
Just as a side note: As far as I know there never was an i2c driver for the saml21 and the effort to unify these drivers has just started. That's why I assume the current driver under sam21_common was never meantt to run for all sam21 variants.

dylad, 2016-10-20:
What would be the best way to integrate i2c driver for saml21 ?
- Add and Modify CMSIS files for saml21 from Atmel ? (I've read this is not a wanted solution)
- Modify i2c.c under sam21_common/periph to integrate specific code for saml21 ?
- Find a way to ignore sam21_common/periph/i2c.c and write a new i2c.c under cpu/saml21/periph ?
Or maybe someone has a better idea ?

miri64, 2016-10-20:
@dylad I think its best to sync in #4926 with the unification effort.

ant9000, 2016-12-01:
Hello,
in my effort to add support for [Acmesystems' YARM](http://acmesystems.it/yarm) I've reworked the I2C driver for SAML21 (the specific chip is L21E18B) - it works correctly as far as my tests go.

I don't really know how to make a pull request cherrypicking only the I2C code, but here's my version anyway:

  https://github.com/ant9000/RIOT/blob/master/cpu/sam21_common/periph/i2c.c

I've added the CMSIS include files, following the idea used for the other sam21 cpus:

  https://github.com/ant9000/RIOT/tree/master/cpu/sam21_common/include/

Ideally, it should be possible to iron out some differences and then get rid completely of `cpu/saml21/include/atmel/`, as was done for D21 and R21.

Antonio

PeterKietzmann, 2016-12-12:
Hi Antonio. Please follow these steps when creating a PR 
[https://github.com/RIOT-OS/RIOT/wiki/Development-procedures](https://github.com/RIOT-OS/RIOT/wiki/Development-procedures). 

Also have a look at 
[https://github.com/RIOT-OS/RIOT/wiki/Contributing-to-RIOT](https://github.com/RIOT-OS/RIOT/wiki/Contributing-to-RIOT) 

and
 [https://github.com/RIOT-OS/RIOT/wiki/Git-cheatsheet](https://github.com/RIOT-OS/RIOT/wiki/Git-cheatsheet)

As @miri64 already stated, you should be aware of  #4926 and synchronize with @haukepetersen.

ant9000, 2016-12-12:
Hi Peter, thanks for the guidelines.

I just tried my hand with a simple PR, that adds support for a SAM L21 cpu variant: #6201 The PR should be quite easy to validate.

 I2C modifications for SAM L21 could follow, but I'll get in touch with @haukepetersen  first.

dylad, 2016-12-12:
Hi,
I give a try for I2C driver on SAML21
Have a look there #5992 
If you have any suggestions about it feel free to comment it.

PeterKietzmann, 2016-12-13:
(How) does #6201 solve your problem? There is still no i2c driver for the saml21. Would you like to have a look at #5992 :-)? However, I don't really see this as an issue. Fine to close this one and further discuss inside above mentioned PRs?

ant9000, 2016-12-13:
@PeterKietzmann , no, my PR does not solve the original issue - sorry, I'm probably just writing in the wrong place.

I've unified the sam21_common driver, in order to make it work on L21 chips - my solution is pretty similar to #5992 , except that I've added the Atmel include files. In order to get rid of the current ones inside cpu/saml21/include/atmel/, though, the current issue with sercom need to be resolved.

dylad, 2016-12-13:
As @PeterKietzmann asked. I closed this issue and further discuss should be move to mentioned PRs.

closed 2016-12-13 by dylad