RIOT-OS/RIOT #5986: overflow in implicit constant conversion

sinkarharshad, 2016-10-23:
I am trying to run a program from RIOT examples "arduino_hello-world" on arduino-uno. 

I am using "make BOARD=arduino-uno flash" this command to run the code given in example. 

and I am getting the following error. 

[image]

Kindly help me to resolve this issue. Thank you in advance.

aabadie, 2016-10-24:
@sinkarharshad, I issued this [PR](#5987) to fix your issue. Feel free to try it and give some feedback.

sinkarharshad, 2016-10-24:
@aabadie Thank you for your efforts sir, saw the issue #5987 : To try this updated changes .. do I need to trash my old RIOT and clone updated RIOT again to my system using "git clone git://github.com/RIOT-OS/RIOT.git" ? 

Thank you in advance.

aabadie, 2016-10-24:
From your local RIOT repository, you can fetch my RIOT remote and checkout in the PR's branch:
```
$ git remote add aabadie git@github.com:aabadie/RIOT.git
$ git fetch aabadie
$ git checkout arduino_sketch_on_arduino
```

sinkarharshad, 2016-10-24:
I tried, and getting following error: 
[image]

miri64, 2016-10-24:
@sinkarharshad use `git://github.com/aabadie/RIOT.git` or `https://github.com/aabadie/RIOT.git` instead.

```bash
git remote set-url aabadie git://github.com/aabadie/RIOT.git
```

sinkarharshad, 2016-10-24:
@aabadie  @miri64  Thank you. 
I followed all the steps, and successfully run the code. But LED's did not flash on uno :! 
What exactly I am missing ): .. Kindly help!! Thank you in advance.
[image]

aabadie, 2016-10-24:
>  But LED's did not flash on uno :! 

Just checked that and in fact it does but too fast. This is because the sleep(500) has no effect. Looks like the xtimer doesn't work with this board.

sinkarharshad, 2016-10-24:
@aabadie Hello Sir, I just observed that after running any program I am getting this 
main(): This is RIOT! (Version: 2016.10-devel-449-g51a0-Harshads-MacBook-Air.local-**arduino_sketch_on_arduino)**

Why there is "arduino_sketch_on_arduino" this thing coming? :! 
Kindly help and thank you in advance.

[image]

taher143, 2016-11-04:
> Looks like the xtimer doesn't work with this board.
how can we use delay for arduino code? timer? what is Xtimer? is it same as Timer0/1/2 in AVR ATmega328P?

OlegHahm, 2016-12-23:
Did anyone open an issue for the xtimer problem?

mali, 2017-01-16:
@OlegHahm : issue #6358 for xtimer problem

PR #5987 (2017-01-24): sys/arduino: make arduino sketch works on avr based boards
This is an attempt to fix #5986.

I made several small changes:
- added the cpp and arduino features in arduino_common boards to avoid a warning message at the beginning of the build
- fixed the implicit conversion warning by using uint32_t instead of int in the RIOT Serial.begin() and sleep() functions of the arduino-like API
- Changed the default baudrate from 115200 to 9600 in the provided example because avr based boards doesn't support 115200 ([see this comment](https://github.com/RIOT-OS/RIOT/blob/master/boards/arduino-common/inc [... cut]
closed 2017-01-24 by haukepetersen