RIOT-OS/RIOT #5990: error: no newline at end of file [-Werror,-Wnewline-eof]

sinkarharshad, 2016-10-24:
I am trying to run a program (attached below). I can run this program in my terminal without any error. But when I am trying to run this simple program in RIOT, getting this following error. Kindly help me to resolve this issue. Thank you in advance.
Figure 1 : Error, Figure 2 : My Code

[image]
 
[image]

Kijewski, 2016-10-24:
Just add a blank line at the end of the file. The standard mandates that a C(++) file needs to have a trailing newline.

sinkarharshad, 2016-10-24:
@Kijewski  Thank you sir. That solved my problem. But after executing my code.. I saw following  strange thing. (Also find the below attached image) 
main(): This is RIOT! (Version: 2016.10-devel-449-g51a0-Harshads-MacBook-Air.local-arduino_sketch_on_arduino)

Why there is "arduino_sketch_on_arduino" this thing coming? :! 
Kindly help and thank you in advance.
[image]

closed 2016-10-25 by sinkarharshad

Kijewski, 2016-10-25:
Actually I wasn't even aware that you can use standard iostreams and threads in RIOT now.

@josephnoir, @gebart, does this look right to you?

sinkarharshad, 2016-10-25:
@Kijewski  :P okay .. anyway .. Thank you for your help.

Kijewski, 2016-10-25:
@sinkarharshad, I didn't mean any disrespect. I only like to learn, too. :) Could be that you can use `<thread>` and `<iostream>` by now. I didn't follow the lastest improvements to RIOT too closely.

sinkarharshad, 2016-10-25:
@Kijewski Hey, I did not feel it that way.. I was really thankful and you actually solved my problem. so I am really grateful :))

josephnoir, 2016-10-25:
It works for the native port, I guess. Trying to cross-compile it for a board should fail:

```
$ BOARD=stm32f4discovery make
Warning: no PORT set!
Building application "riot_and_cpp" for "stm32f4discovery" with MCU "stm32f4".

"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C ~/RIOT/boards/stm32f4discovery
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C ~/RIOT/core
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C ~/RIOT/cpu/stm32f4
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C ~/RIOT/drivers
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C ~/RIOT/cpu/cortexm_common
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C ~/RIOT/drivers/periph_common
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C ~/RIOT/cpu/stm32_common
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C ~/RIOT/sys
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C ~/RIOT/cpu/stm32_common/periph
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C ~/RIOT/cpu/stm32f4/periph
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C ~/RIOT/sys/auto_init
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C ~/RIOT/sys/newlib
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C ~/RIOT/sys/tsrb
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C ~/RIOT/sys/uart_stdio
~/RIOT/examples/riot_and_cpp/main.cpp: In function 'int main()':
~/RIOT/examples/riot_and_cpp/main.cpp:16:3: error: 'thread' was not declared in this scope
   thread t1((Fctor()));
   ^
~/RIOT/examples/riot_and_cpp/main.cpp:20:3: error: 't1' was not declared in this scope
   t1.join();
   ^
make[1]: *** [~/RIOT/examples/riot_and_cpp/bin/stm32f4discovery/riot_and_cpp/main.o] Error 1
make: *** [all] Error 2
```

To allow using something very similar std::thread on RIOT, there is the [cpp11-compat](https://github.com/RIOT-OS/RIOT/tree/master/sys/cpp11-compat) module. Since there is no C++ clock on RIOT, functions that require a timer accept a custom struct, see the [riot/chrono.hpp](https://github.com/RIOT-OS/RIOT/blob/master/sys/cpp11-compat/include/riot/chrono.hpp).

KatiePlayz, 2021-10-30:
https://www.youtube.com/watch?v=Si8rN5J249M
i had the same problemo and i fixed