RIOT-OS/RIOT #6017: Thread Synchronisation: Mutex

sinkarharshad, 2016-10-29:
I am trying to synchronise two threads (t1 and main) using mutex function (C++). There are no errors or any warnings (tried and tested in C++, Worked fine). But when I am trying to run the same code in RIOT I am getting following error as attached below. (Figure1: Error, Figure2: My code). 
Kindly help me to resolve this issue. Thank you in advance!! 

[image]
[image]

josephnoir, 2016-10-29:
I tried to reproduce this and it works as intended for me using the native port:

```bash
$ make
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/c++/4.2.1
Building application "riot_and_cpp" for "native" with MCU "native".

"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C ~/Git/RIOT/boards/native
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C ~/Git/RIOT/core
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C ~/Git/RIOT/cpu/native
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C ~/Git/RIOT/boards/native/drivers
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C ~/Git/RIOT/drivers
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C ~/Git/RIOT/cpu/native/periph
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C ~/Git/RIOT/sys
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C ~/Git/RIOT/sys/auto_init
__TEXT  __DATA  __OBJC  others  dec hex
32768 806912  0 20480 860160  d2000 
bash: objcopy: command not found
make: *** [all] Error 127
$ make term
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/c++/4.2.1
~/Git/RIOT/examples/riot_and_cpp/bin/native/riot_and_cpp.elf  
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2014.01-10146-gb2bed-computer.local)
From main: 0
From t1: 0
From main: 1
From t1: 1
From main: 2
From t1: 2
From main: 3
From t1: 3
From main: 4
From t1: 4
From main: 5
From t1: 5
From main: 6
From t1: 6
From main: 7
From t1: 7
From main: 8
From t1: 8
From main: 9
From t1: 9
^Clpm_set(): exit()

$
```

Here is my code:

```C++
#include <iostream>
#include <thread>
#include <string>
#include <mutex>

using namespace std;

mutex mu;

void shared_print(const string& msg, int id) {
  mu.lock();
  cout << msg << id << endl;
  mu.unlock();
}

void function_1() {
  for (int i = 0; i < 10; ++i) {
    shared_print("From t1: ", i);
  }
}

int main() {
  thread t1(function_1);

  for (int i = 0; i < 10; ++i) {
    shared_print("From main: ", i);
  }
  t1.join();
}
```

Note that compiling C++ code for a board won't work with the STL thread and mutex headers. You can use the [c++ 11 compat](https://github.com/RIOT-OS/RIOT/tree/master/sys/cpp11-compat) module which offers thread, mutex and condition variable headers C++11 (excluding timer dependent functionality).

So, what compiler did you use? Is there anything else configured in your build environment?

Btw, it would be nice if you could provide code example using [GitHub Markdown](https://help.github.com/articles/basic-writing-and-formatting-syntax/#quoting-code) to make it easier to work with!

sinkarharshad, 2016-10-29:
@josephnoir Thank you for your feedback.
1: I tried to run your program - Worked fine in C++  - Got same error in RIOT :!
2: I am using a macos-terminal to compile my programs.

Where m I going wrong? 
Kindly help!!

Thank you in advance.

sinkarharshad, 2016-10-29:
I did it. Thank you for your help :))

closed 2016-10-29 by miri64