RIOT-OS/RIOT #6018: nRF52 gnrc 6lowpan ble memory leak

jfbortolotti, 2016-10-29:
In case of Tx conflict on the ble interface, the _send function of gnrc_nordic_ble_6lowpan.c will return an -EBUSY error but no one handling the GNRC_NETAPI_MSG_TYPE_SND will take care of the error and what needs to be done for the packet(free or requeue).
This packet will stay in the alloc buffer.
After some time, if such conflicts happened, the memory allocated will continue to grow till the gnrc packet buffer will be full.
This can be reproduced with the microcoap_server example and sending pings forever. At some point, the buffer will be full and the stack will not be able to handle the ping echo request and reply at all.

One easy workaround is to accept to loose packet when there is such conflict and freeing up the packet.
In https://github.com/RIOT-OS/RIOT/blob/master/pkg/nordic_softdevice_ble/src/gnrc_nordic_ble_6lowpan.c#L139 
add a call to release the gnrc packet buffer:
    if (ble_mac_busy_tx) {
        DEBUG("_send(): ble_mac_busy_tx, dropping packet\n");
        gnrc_pktbuf_release(pkt);
        return -EBUSY;
    }

With such "ugly" workaround, the memory leak will be dramatically reduced and the stack will be able to handle more than 5000 pings without memory allocation issue. 
There is simlar memory leak in case of -EBADMSG : https://github.com/RIOT-OS/RIOT/blob/master/pkg/nordic_softdevice_ble/src/gnrc_nordic_ble_6lowpan.c#L149

One better approach will be to requeue the GNRC_NETAPI_MSG_TYPE_SND  message for retransmission. This could be done either the _send function or the _gnrc_nordic_ble_6lowpan_thread main loop by handling the different errors returned by _send.

Looking for feedback on the recommendation for properly fixing this.

Note:
1/ to get ping stability, you will need a fix or workaround regarding the uart_write function depending on how much debug you have activated. See #5954 .
2/ there is still a remaining memory leak after this, not yet understood by with a lot lower magnitude because it is only 8bytes once in a while and there is usually an hard_fault  happening before the memory allocation buffer is full.

kaspar030, 2016-11-18:
Sorry for the delay.

> One better approach will be to requeue the GNRC_NETAPI_MSG_TYPE_SND message for retransmission. This could be done either the _send function or the _gnrc_nordic_ble_6lowpan_thread main loop by handling the different errors returned by _send.

IMHO send would be the place to handle the error. I'd be fine with a workaround (make sure the packet always gets freed), but re-queuing is maybe not much more complicated to implement?

jfbortolotti, 2016-11-30:
Correct re-queuing may not be that complicated it will just take more time for me to figure it out and test it ...
If you can point me to some sample of code, I'll give it a try.
I can submit the workaround very quickly as I tested it already.

kaspar030, 2016-11-30:
> I can submit the workaround very quickly as I tested it already.

Then let's go with the workaround and provide a proper fix later!

kaspar030, 2016-11-30:
One more idea: could we replace ```ble_mac_busy_tx``` with a mutex?

kYc0o, 2017-04-25:
So, there was such workaround?

jfbortolotti, 2017-04-25:
Nothing cleaner than what described above so far.

kYc0o, 2017-04-26:
What I mean is if a PR was opened with such workaround.

jfbortolotti, 2017-04-26:
No PR opened with such workaround.

kYc0o, 2018-07-17:
@pekkanikander your PR updating the softdevice would solve this issue?

pekkanikander, 2018-07-21:
> @pekkanikander your PR updating the softdevice would solve this issue?

I don't know.  I haven't had time to look at this bug.

miri64, 2019-09-23:
As nimble is now used per default with the NRF52dk instead of the soft device, shouldn't this be obsolete now?

aabadie, 2019-09-23:
> As nimble is now used per default with the NRF52dk instead of the soft device, shouldn't this be obsolete now?

SoftDevice is still supported in RIOT so this is bug is still there. Maybe just update the PR title ? And try to apply the proposed fix ?

miri64, 2019-09-23:
> […] And try to apply the proposed fix ?

Then please do, I don't believe there is anyone working with it right now though, due to several porting issues.

Citrullin, 2020-02-09:
> > As nimble is now used per default with the NRF52dk instead of the soft device, shouldn't this be obsolete now?
> 
> SoftDevice is still supported in RIOT so this is bug is still there. Maybe just update the PR title ? And try to apply the proposed fix ?

I agree with @miri64. The package is not well maintained anymore, because NimBLE provides a better job and is the default package for nrf52 ble related things. We might even want to discuss removing the softdevice in the long run. Migration should be really simple due to the gnrc network stack.

miri64, 2020-07-01:
The softdevice is slated to be removed after 2020.10. If someone wants to fix that bug until then, it is very welcome, but for now I just assign the milestone as a memo for the next release manager.

chrysn, 2021-05-10:
With the removal of the softdevice complete, this has become moot; closing.

closed 2021-05-10 by chrysn