RIOT-OS/RIOT #6027: How to get process ID in RIOT?

sinkarharshad, 2016-10-31:
I am trying to get the process ID in RIOT. But not able to do it and getting the following attached error :! . Same program is working fine in C++. 
[image]

My code is as follows:
#include <iostream>
#include <sys/types.h>
#include <unistd.h>
#include <lpm.h>

using namespace std;

int main ()
{
	pid_t pidorzero = fork ();
	int x = 1;
	
	if(pidorzero < 0)
	{
		perror("fork () error");
	}

	if (pidorzero > 0)
	{
		cout << "I am the Parent Process with x: " << x << endl;
		cout << "ParentPID: " << getpid ()<< endl;	
	}

	else
	{
		cout << "I am the Child Process with x: " << x << endl;
		cout << "ChildPID: " << getpid() << endl;	
	}
	lpm_set(LPM_OFF);
	return 0;
}

miri64, 2016-10-31:
(I allowed myself to edit your post to make the code more readable)

`fork()` is not implemented in RIOT yet and it is highly unlikely that it ever will be (every new process would need a memory stack which costs memory of which we only have little ;-)). Also consider to post your questions to the [devel@riot-os.org](https://lists.riot-os.org/mailman/listinfo/devel) or the [users@riot-os.org](https://lists.riot-os.org/mailman/listinfo/users) mailing list or the [IRC chat (chat.freenode.org #riot-os)](http://webchat.freenode.net?channels=%23riot-os). You are usually reaching a bigger, more diverse group of people there. The issue tracker is primarily used for reporting bugs so you will primarily find developers here ;-).

miri64, 2016-10-31:
(if you want to get the process ID of your `native` instance: it is accessable through the [CPUID driver](http://doc.riot-os.org/group__drivers__periph__cpuid.html))

closed 2016-10-31 by miri64