RIOT-OS/RIOT #6035: How to program Standalone ATmega328P

taher143, 2016-11-01:
How can I port a code into stand alone ATmega328P board instead of Arduino? I do need to change avrdude command to program my ATmega328P. From where can I change this command?

mali, 2016-11-28:
the avrdude command used to flash arduino boards is defined here : https://github.com/RIOT-OS/RIOT/blob/master/boards/arduino-common/Makefile.include 
arduino-common refers to uno and duemilanove which are the arduino boards based on atmega328p mcu. 
Hope this helps

OlegHahm, 2016-12-08:
@taher143, did this answer your question?

smlng, 2017-07-06:
is this solved - it also seems more like a flash-tool question than RIOT specific, anyway.

aabadie, 2018-04-14:
Closing this one since there's no activity and an answer was proposed.

Feel free to reopen if you disagree.

closed 2018-04-14 by aabadie