RIOT-OS/RIOT #6036: tinydtls: DTLS randomly missing functions

miri64, 2016-11-01:
Murdock builds are randomly failing because sometimes (but not always) there seem to be a linking problem with DTLS. An example for this can be seen in [the build for #6009](https://ci.riot-labs.de/RIOT-OS/RIOT/6009/0dfcefc003171dcded7833f9d9ebe0cd1a3f20f1/output.html#arm7):

```
dtls-echo:pttu:
Building application "dtls_echo" for "pttu" with MCU "lpc2387".

make -C /data/riotbuild/examples/dtls-echo/bin/pkg/pttu/tinydtls
/data/riotbuild/examples/dtls-echo/bin/pttu/tinydtls.a(crypto.o): In function `dtls_encrypt':
/data/riotbuild/examples/dtls-echo/bin/pkg/pttu/tinydtls/crypto.c:527: undefined reference to `rijndael_set_key_enc_only'
/data/riotbuild/examples/dtls-echo/bin/pttu/tinydtls.a(crypto.o): In function `dtls_decrypt':
/data/riotbuild/examples/dtls-echo/bin/pkg/pttu/tinydtls/crypto.c:553: undefined reference to `rijndael_set_key_enc_only'
/data/riotbuild/examples/dtls-echo/bin/pttu/tinydtls.a(ccm.o): In function `mac':
/data/riotbuild/examples/dtls-echo/bin/pkg/pttu/tinydtls/ccm.c:163: undefined reference to `rijndael_encrypt'
/data/riotbuild/examples/dtls-echo/bin/pttu/tinydtls.a(ccm.o): In function `encrypt':
/data/riotbuild/examples/dtls-echo/bin/pkg/pttu/tinydtls/ccm.c:149: undefined reference to `rijndael_encrypt'
/data/riotbuild/examples/dtls-echo/bin/pttu/tinydtls.a(ccm.o): In function `add_auth_data':
/data/riotbuild/examples/dtls-echo/bin/pkg/pttu/tinydtls/ccm.c:81: undefined reference to `rijndael_encrypt'
/data/riotbuild/examples/dtls-echo/bin/pkg/pttu/tinydtls/ccm.c:121: undefined reference to `rijndael_encrypt'
/data/riotbuild/examples/dtls-echo/bin/pkg/pttu/tinydtls/ccm.c:128: undefined reference to `rijndael_encrypt'
/data/riotbuild/examples/dtls-echo/bin/pttu/tinydtls.a(ccm.o):/data/riotbuild/examples/dtls-echo/bin/pkg/pttu/tinydtls/ccm.c:136: more undefined references to `rijndael_encrypt' follow
collect2: error: ld returned 1 exit status
/data/riotbuild/Makefile.include:262: recipe for target 'all' failed
make: *** [all] Error 1
/data/riotbuild/Makefile.buildtests:38: recipe for target 'buildtest' failed
```

Could not assign @rfuentess so I assigned @OlegHahm.

OlegHahm, 2016-11-01:
Hm, I seem to remember to have seen this problem before - but don't remember if I've ever understood the reason for that.

rfuentess, 2016-11-01:
yes, I had watched this issue when tested the dtls-echo with `make buildtests` but frankly I though was something more related to RIOT than my code.

kYc0o, 2016-11-01:
As I commented on that PR, this error appears systematically on OS X, and as far as I can remember, it changes randomly.

miri64, 2016-11-04:
Fixed in #6053.

closed 2016-11-04 by miri64