RIOT-OS/RIOT #6037: pkg/nanocoap: native build fails on macOS

smlng, 2016-11-01:
Building `examples/nanocoap_server` or the (new) `examples/gcoap` fails on macOS with:

```
/RIOT/examples/nanocoap_server/bin/pkg/native/nanocoap/nanocoap/nanocoap_sock.c:91:32: error: 
      missing field 'addr' initializer [-Werror,-Wmissing-field-initializers]
    sock_udp_ep_t remote = { 0 };
                               ^
1 error generated.
```

Further, there is an error when the user has no `~/.gitcache` directory.

miri64, 2016-11-01:
> [EDIT]: though it has something to do with the new sock API, tests/gnrc_sock_udp/ builds on macOS for native.

Because I used field-initializers there ;-). But darn, I wanted to use the `{ 0 }` notation inmy current ZEP port.

Can you try if

```C
sock_udp_ep_t remote = { .family = AF_INET };
```

does the trick?

smlng, 2016-11-02:
>Can you try if
>```
>sock_udp_ep_t remote = { .family = AF_INET };
>```
>does the trick?
@miri64: works! I'll open issue in @kaspar030 sock/nanocoap reop.

smlng, 2016-12-09:
solved with #6101

closed 2016-12-09 by smlng