RIOT-OS/RIOT #6043: iotlab-m3 nodes always fail on flashing

kYc0o, 2016-11-02:
While testing the release specs I found that the iotlab-m3 is flashed correctly, but I always have an error.

Can anyone else confirm this behaviour on ` 2016.10-RC1`?

My config is:

```console
Open On-Chip Debugger 0.9.0-dev-gbdd8312 (2016-01-12-18:06)
```

And

```console
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.2.1 20151202 (release) [ARM/embedded-5-branch revision 231848]
```

I'm in a Debian jessie machine this time ;-)

thomaseichinger, 2016-11-02:
@kYc0o What is the actual error (message)?

kYc0o, 2016-11-02:
Sorry I forgot to put it in the description:

```
/home/facosta/git/RIOT-OS/RIOT/tests/emb6/../../Makefile.include:329: recipe for target 'flash' failed
make: *** [flash] Error 1
```

miri64, 2016-11-03:
Tat was one of my main issues, when trying to implement a working test framework ;-)

kYc0o, 2016-11-07:
Just tested on OS X with `Open On-Chip Debugger 0.10.0-dev-gdfc6658 (2016-02-18-22:16)` and it works as expected.

OlegHahm, 2016-11-08:
The reason is probably the old OpenOCD version. See discussion in https://github.com/RIOT-OS/RIOT/pull/2472.

kYc0o, 2016-11-08:
Yes, that's the problem. We can then say somewhere (wiki?) that we recommend OpenOCD 0.10 or higher?

OlegHahm, 2016-11-08:
Actually, I thought we already did this, but I cannot find it. So, :+1: for recommending OpenOCD > 0.9 somewhere.

kYc0o, 2016-11-10:
Done in the [wiki](https://github.com/RIOT-OS/RIOT/wiki/OpenOCD).

closed 2016-11-10 by kYc0o

Joe-sae, 2017-08-03:
I have install RIOT and openocd followed wed size https://github.com/RIOT-OS/RIOT/wiki/OpenOCD and https://github.com/RIOT-OS/RIOT/wiki.
anything is OK. 
But it is fail to make flash after I connect ST-LINK with STM32 broad,.
it shows that  Error: corrupted fifo read pointer 0xe7fee7fe.
could you give me some help ?

----------------------------------------------
next information is catching in Ubuntu.

$ make flash
Building application "test_stm32" for "nucleo-f103" with MCU "stm32f1".

"make" -C /home/joe/win_share/RIOT-master/boards/nucleo-f103
"make" -C /home/joe/win_share/RIOT-master/core
"make" -C /home/joe/win_share/RIOT-master/cpu/stm32f1
"make" -C /home/joe/win_share/RIOT-master/cpu/cortexm_common
"make" -C /home/joe/win_share/RIOT-master/cpu/cortexm_common/periph
"make" -C /home/joe/win_share/RIOT-master/cpu/stm32_common
"make" -C /home/joe/win_share/RIOT-master/cpu/stm32_common/periph
"make" -C /home/joe/win_share/RIOT-master/cpu/stm32f1/periph
"make" -C /home/joe/win_share/RIOT-master/drivers
"make" -C /home/joe/win_share/RIOT-master/drivers/periph_common
"make" -C /home/joe/win_share/RIOT-master/sys
"make" -C /home/joe/win_share/RIOT-master/sys/auto_init
"make" -C /home/joe/win_share/RIOT-master/sys/isrpipe
"make" -C /home/joe/win_share/RIOT-master/sys/newlib_syscalls_default
"make" -C /home/joe/win_share/RIOT-master/sys/pm_layered
"make" -C /home/joe/win_share/RIOT-master/sys/tsrb
"make" -C /home/joe/win_share/RIOT-master/sys/uart_stdio
   text	   data	    bss	    dec	    hex	filename
   8492	    136	   2756	  11384	   2c78	/home/joe/win_share/RIOT-master/examples/test_stm32/bin/nucleo-f103/test_stm32.elf
/home/joe/win_share/RIOT-master/dist/tools/openocd/openocd.sh flash
### Flashing Target ###
Open On-Chip Debugger 0.10.0+dev-00169-g02bc718 (2017-08-01-13:38)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v27 API v2 SWIM v6 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.256458
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32f1x.cpu       hla_target little stm32f1x.cpu       halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080003f4 msp: 0x20000200
auto erase enabled
Info : device id = 0x10036414
Info : flash size = 512kbytes
Error: corrupted fifo read pointer 0xe7fee7fe
Error: timed out while waiting for target halted
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x61000003 pc: 0x08001040 msp: 0x20000180
Error: error waiting for target flash write algorithm
Error: error writing to flash at address 0x08000000 at offset 0x00000000

/home/joe/win_share/RIOT-master/examples/test_stm32/../../Makefile.include:370: recipe for target 'flash' failed
make: *** [flash] Error 1