RIOT-OS/RIOT #6052: tests: xtimer_drift isn't working properly

miri64, 2016-11-03:
It was shown in https://github.com/RIOT-OS/Release-Specs/issues/27 that the xtimer_drift currently stops at about ~30-90 seconds. I'm not sure if this is related to #5103. In trying to bisect this I reallized that the same thing might also happen in earlier versions but way later (up to 590 seconds I saw).

A-Paul, 2016-11-10:
Just for the record. While testing for 2016.10 RC4 the test has stopped after ~7min, ~3min and ~20min.

miri64, 2016-11-11:
Out of interest I tried this on 2016.10-devel, too: it's also freezing after ~3 min.

miri64, 2016-11-11:
On its introduction the application worked. I leave a longterm bisect run over the weekend ;-)

smlng, 2017-01-31:
I just ran the `xtimer_drift` test on native several times for the `2017.01-RC2` release. It got stuck after 62, 681, and 272 seconds for instance. So not deterministic, however, I did not dig any deeper into this.

miri64, 2017-01-31:
(btw my longterm bisect was as inconclusive as it gets...)

smlng, 2017-02-01:
I narrowed it down to the assembly code [here](https://github.com/RIOT-OS/RIOT/blob/master/cpu/x86/x86_ucontext.c#L60), that is the last function call I see every time before it gets stuck.

My assembly magic is very low, so I didn't dig any deeper ...

kYc0o, 2017-04-25:
I confirm this bug, but I have no idea on the cause... any more hints @smlng ?

smlng, 2017-04-25:
@kYc0o didn't look into this any further, so no 😞

haukepetersen, 2017-10-24:
Just ran the test again for the `2017.10` release spec tests on native: same faulty behavior - tests gets stuck after random intervals.

MichelRottleuthner, 2020-01-24:
Just for reference: in current master (3c13df60abe22e544840a8b9befb39ad3d3bb391) this Issue is still present. Also I'll add the native label as it is only reproducible on native.

maribu, 2022-09-16:
With xtimer on its way out and ztimer being used instead now every, I'd say this is a wontfix.

If anyone disagrees, please reopen.

closed 2022-09-16 by maribu