RIOT-OS/RIOT #6056: Compiler msp430-gcc is required but not found in PATH.  Aborting.

sinkarharshad, 2016-11-03:
I am trying port RIOT on Texas Instruments EZ430-Chronos. and getting this error. 
Kindly help me to resolve this issue.
Thank you in advance. 
(I am using mac :!)

miri64, 2016-11-04:
You don't have to port the `chronos`, there is [already a port for that](https://github.com/RIOT-OS/RIOT/tree/master/boards/chronos). You have to install the compiler for MSP430 (the architecture the `chronos` is based on). @emmanuelsearch @kYc0o @Lotterleben @smlng @thomaseichinger @Yonezawa-T2 maybe you can help how to do that?

Also I again urge you to ask this kind of questions to the [devel](https://lists.riot-os.org/mailman/listinfo/devel) mailing [list](https://en.wikipedia.org/wiki/Electronic_mailing_list). People there are as eager to help you there as they are here, with the bonus that you reach more people (not just maintainers and some stray contributors stumbling at random over this issue)! :-)

sinkarharshad, 2016-11-04:
@miri64 I solved it in linux .. but thank you :)

kYc0o, 2016-11-07:
It's solved so I'll close it.

closed 2016-11-07 by kYc0o