RIOT-OS/RIOT #6068: Failed assertion on z1 while receiving a packet

kYc0o, 2016-11-07:
While testing the default example in current master I noticed a failed assertion on z1 hardware when a packet is received:

```console
2016-11-07 17:38:58,930 - INFO # > n/a
2016-11-07 17:38:58,930 - INFO # *** RIOT kernel panic:
2016-11-07 17:38:58,931 - INFO # FAILED ASSERTION.
2016-11-07 17:38:58,931 - INFO # 
2016-11-07 17:38:58,933 - INFO # 	pid | name                 | state    Q | pri | stack ( used) | base       | current    
2016-11-07 17:38:58,935 - INFO # 	  - | isr_stack            | -        - |   - |   256 (   -1) |     0xffff |     0xffff
2016-11-07 17:38:58,937 - INFO # 	  1 | idle                 | pending  Q |  15 |    96 (   60) |     0x1126 |     0x114a 
2016-11-07 17:38:58,940 - INFO # 	  2 | main                 | bl mutex _ |   7 |   512 (  286) |     0x1186 |     0x1298 
2016-11-07 17:38:58,944 - INFO # 	  3 | pktdump              | bl rx    _ |   6 |   512 (  154) |     0x197c |     0x1ae8 
2016-11-07 17:38:58,946 - INFO # 	  4 | cc2420               | running  Q |   2 |   512 (  368) |     0x1570 |     0x16d6 
2016-11-07 17:38:58,948 - INFO # 	    | SUM                  |            |     |  1888 (  868)
2016-11-07 17:38:58,948 - INFO # 
2016-11-07 17:38:58,948 - INFO # *** halted.
2016-11-07 17:38:58,948 - INFO # 
2016-11-07 17:38:58,948 - INFO # 
2016-11-07 17:38:58,996 - INFO # RIOT MSP430 hardware initialization complete.
2016-11-07 17:38:58,996 - INFO # 
2016-11-07 17:38:59,008 - INFO # main(): This is RIOT! (Version: 2017.01-devel-26-ge6ad43-bluelatex.saclay.inria.fr)
2016-11-07 17:38:59,009 - INFO # Welcome to RIOT!
```

The packet comes from a samr21-xpro, also using the default example.

kYc0o, 2016-11-07:
Just found that the failing assertion is [here](https://github.com/RIOT-OS/RIOT/blob/6cd8d98f1d9638f697ce9eadc1a3420bc21cebb4/cpu/msp430fxyz/periph/spi.c#L221) but only when receiving packets. If I comment out this line everything works as expected.

Any ideas @haukepetersen ?

kYc0o, 2016-11-07:
Another strange thing: Nothing fails on gnrc_minimal on z1 side, pings are answered correctly from a samr21-xpro running gnrc_networking. Until now it only fails when testing the default application on the z1.

thomaseichinger, 2016-11-07:
@kYc0o Could you please test if #6070 fixes this issue?

kYc0o, 2016-11-08:
Fixed by #6070 .

closed 2016-11-08 by kYc0o