RIOT-OS/RIOT #6073: #5745 breaks my board

roberthartung, 2016-11-08:
Hey everyone,

I just pulled the current master into my fork and it turns out that I am not able to run the default example anymore. If I revert anything related to #5745 it works again.

It looks like the pktdump msg_queue is not initialized for some reason. The gnrc_netreg_register() function is called for the pktdump thread without it being initialized!

```
2016-11-08 08:38:58,770 - INFO # n/a
2016-11-08 08:38:58,815 - INFO # *** RIOT kernel panic in /home/hartung/git/RIOT/sys/net/gnrc/netreg/gnrc_netreg.c:41:
2016-11-08 08:38:58,824 - INFO # FAILED ASSERTION.
2016-11-08 08:38:58,825 - INFO # 
2016-11-08 08:38:58,871 - INFO # 	pid | name                 | state    Q | pri | stack ( used) | base       | current    
2016-11-08 08:38:58,917 - INFO # 	  - | isr_stack            | -        - |   - |   128 (   -1) |     0xffff |     0xffff
2016-11-08 08:38:58,964 - INFO # 	  1 | idle                 | pending  Q |  15 |   128 (   66) |     0x1a96 |     0x1ad3 
2016-11-08 08:38:59,011 - INFO # 	  2 | main                 | running  Q |   7 |   384 (  316) |     0x1b16 |     0x1c53 
2016-11-08 08:38:59,057 - INFO # 	  3 | pktdump              | pending  Q |   6 |   384 (   66) |     0x1d0a |     0x1e47 
2016-11-08 08:38:59,104 - INFO # 	  4 | at86rf2xx            | pending  Q |   2 |   256 (   66) |     0x1ecf |     0x1f8c 
2016-11-08 08:38:59,137 - INFO # 	    | SUM                  |            |     |  1280 (  514)
2016-11-08 08:38:59,138 - INFO # 
2016-11-08 08:38:59,144 - INFO # *** halted.
```

If anyone could look into this it would be great!

Best Regards,
Robert

miri64, 2016-11-08:
That is not a bug, it's a failed assertion. The message queue of a registering thread needs to be initialized when running GNRC. I don't see how this is related to #5745.

OlegHahm, 2016-11-08:
If the default example fails with an assertion, I would call it a bug.

roberthartung, 2016-11-08:
It is clearly a bug. ... or at least the changes from #5745 make the assertion appear! If I simply comment the changes from that issue the both my program and the default example are working again!

I assume that scheduling happens too early or something else.

I added a printf to `sys/net/gnrc/pktdump/gnrc_pktdump.c:168` (just inside the `_eventloop` function). And if I don't use the changes from #5745 I get correct output:

```
2016-11-08 11:37:08,206 - INFO # Created thread idle. PID: 1. Priority: 15.
2016-11-08 11:37:08,227 - INFO # Created thread main. PID: 2. Priority: 7.
2016-11-08 11:37:08,237 - INFO # start_threading()
2016-11-08 11:37:08,260 - INFO # Created thread pktdump. PID: 3. Priority: 6.
2016-11-08 11:37:08,271 - INFO # thread_arch_yield()
2016-11-08 11:37:08,277 - INFO # _eventloop..
2016-11-08 11:37:08,288 - INFO # thread_arch_yield()
2016-11-08 11:37:08,307 - INFO # spi.c: dev = 0, conf = 0, speed = 0
2016-11-08 11:37:08,331 - INFO # Created thread at86rf2xx. PID: 4. Priority: 2.
2016-11-08 11:37:08,341 - INFO # thread_arch_yield()
2016-11-08 11:37:08,352 - INFO # thread_arch_yield()
2016-11-08 11:37:08,366 - INFO # thread_arch_yield()
2016-11-08 11:37:08,409 - INFO # main(): This is RIOT! (Version: 2016.10-devel-429-ga0dc2-rhpc-hartung/boards/inga)
2016-11-08 11:37:08,418 - INFO # Welcome to RIOT!
2016-11-08 11:37:08,429 - INFO # > thread_arch_yield()
```

But if I use the code I get:

```
Created thread idle. PID: 1. Priority: 15.
2016-11-08 11:35:47,992 - INFO # Created thread main. PID: 2. Priority: 7.
2016-11-08 11:35:48,001 - INFO # start_threading()
2016-11-08 11:35:48,024 - INFO # Created thread pktdump. PID: 3. Priority: 6.
2016-11-08 11:35:48,034 - INFO # thread_arch_yield()
2016-11-08 11:35:48,053 - INFO # spi.c: dev = 0, conf = 0, speed = 0
2016-11-08 11:35:48,078 - INFO # Created thread at86rf2xx. PID: 4. Priority: 2.
2016-11-08 11:35:48,088 - INFO # thread_arch_yield()
2016-11-08 11:35:48,131 - INFO # main(): This is RIOT! (Version: 2016.10-devel-429-ga0dc2-rhpc-hartung/boards/inga)
2016-11-08 11:35:48,133 - INFO # n/a
2016-11-08 11:35:48,179 - INFO # *** RIOT kernel panic in /home/hartung/git/RIOT/sys/net/gnrc/netreg/gnrc_netreg.c:41:
2016-11-08 11:35:48,188 - INFO # FAILED ASSERTION.
2016-11-08 11:35:48,188 - INFO # 
2016-11-08 11:35:48,235 - INFO # 	pid | name                 | state    Q | pri | stack ( used) | base       | current    
2016-11-08 11:35:48,281 - INFO # 	  - | isr_stack            | -        - |   - |   128 (   -1) |     0xffff |     0xffff
2016-11-08 11:35:48,328 - INFO # 	  1 | idle                 | pending  Q |  15 |   128 (   66) |     0x1aa6 |     0x1ae3 
2016-11-08 11:35:48,375 - INFO # 	  2 | main                 | running  Q |   7 |   384 (  312) |     0x1b26 |     0x1c63 
2016-11-08 11:35:48,422 - INFO # 	  3 | pktdump              | pending  Q |   6 |   384 (   66) |     0x1d1a |     0x1e57 
2016-11-08 11:35:48,469 - INFO # 	  4 | at86rf2xx            | pending  Q |   2 |   256 (   66) |     0x1edf |     0x1f9c 
2016-11-08 11:35:48,502 - INFO # 	    | SUM                  |            |     |  1280 (  510)
2016-11-08 11:35:48,502 - INFO # 
2016-11-08 11:35:48,508 - INFO # *** halted.
2016-11-08 11:35:48,509 - INFO # 
```

miri64, 2016-11-08:
> If the default example fails with an assertion, I would call it a bug.

It's a bug in the application, yes. But not an AVR specific bug and not a bug in the system. Just one look into the default example would show you that the message queue isn't even initialized ;-). The assert @roberthartung hits was introduced in https://github.com/RIOT-OS/RIOT/commit/2e342c7e

miri64, 2016-11-08:
@roberthartung it might be that fixing the threading behavior revealed this shortcomming of the default app on AVR boards.

miri64, 2016-11-08:
Also what changes from #5745 do you mean? It is an issue not a PR, so no changes were made there.

OlegHahm, 2016-11-08:
> It's a bug in the application, yes. But not an AVR specific bug and not a bug in the system. Just one look into the default example would show you that the message queue isn't even initialized ;-).

If you were right the example would fail for every platform. But fortunately this isn't the case, because the the message queue is initialized in `sys/net/gnrc/pktdump/gnrc_pktdump.c`.

OlegHahm, 2016-11-08:
@miri64 wrote:
> Also what changes from #5745 do you mean? It is an issue not a PR, so no changes were made there.

@roberthartung wrote:
> If I revert anything related to #5745 it works again.

The only PR linked in #5745 is #5763.

roberthartung, 2016-11-08:
@OlegHahm You're totally right here. It is NOT a bug in the application! It is a bug of the "fixed" threading for ATmega CPUs / AVR boards.

@miri64 Sorry, but you're wrong. It's a specific problem of avr boards with the changes within that issue / the references pull request.

I am indeed talking about commit https://github.com/RIOT-OS/RIOT/pull/5763/commits/e0365e0bf95b18eec57c8b0808e9984133bd2b83 . If I revert these changes the example works again. As I wrote above the message queue is correctly initialized in `_eventloop` where the printf happens when everything is fine. For some reason the thread is not started/initialized with the changes made to the threading anymore, resulting in the message queue not being initialized!

miri64, 2016-11-08:
Ah okay, sorry I was a little bit misled by the failed assertion >.<.

kYc0o, 2016-11-08:
Can you describe your setup? It seems to me that something is strange, since in our tests we never put an at86rf2xx transceiver together with any AVR cpu while merging such PR, so that's maybe why this specific commit is triggering the failing assert.

roberthartung, 2016-11-08:
@kYc0o I am using one of our INGA sensor nodes ( https://www.ibr.cs.tu-bs.de/projects/inga/index.html?lang=en ).

kYc0o, 2016-11-08:
Just tested on arduino-mega2560 and everything works. Can you provide a link to your fork which is failing? I assume you provided these needed defines:

```c
/**
 + * Context swap defines
 + * Setup to use PJ6 which is pin change interrupt 15 (PCINT15)
 + * This emulates a software triggered interrupt
 + **/
 +#define AVR_CONTEXT_SWAP_INIT do { \
 +    DDRJ |= (1 << PJ6); \
 +    PCICR |= (1 << PCIE1); \
 +    PCMSK1 |= (1 << PCINT15); \
 +} while (0)
 +#define AVR_CONTEXT_SWAP_INTERRUPT_VECT  PCINT1_vect
 +#define AVR_CONTEXT_SWAP_TRIGGER   PORTJ ^= (1 << PJ6)
```

of course by choosing an unused interrupt pin on your custom board.

roberthartung, 2016-11-08:
@kYc0o See https://github.com/roberthartung/RIOT/tree/hartung/boards/inga for the fork. I used the defines from another atmega board, but I've just seen, that they might be just wrong ...Let me check them and I'll come back to you

roberthartung, 2016-11-08:
@kYc0o Ok .. it looks like this was just a misconfiguration on my side ... I will investigate further tomorrow.

kYc0o, 2016-11-10:
@roberthartung Can we close this issue?

roberthartung, 2016-11-11:
@kYc0o it looks like it's working for now! Thanks for the support!

closed 2016-11-11 by roberthartung