RIOT-OS/RIOT #6076: Behaviour of gnrc_sixlowpan_ctx_lookup_addr and IPHC

melshuber, 2016-11-08:
assume a 6ctx table

```
# 0 prefix: fd2a::9a7e:a8e7:3bff:fe9c:9a7c/128, ltime: 29min flags: C=1
# 1 prefix: fd2a::21f:ff:fe00:0/128, ltime: 29min flags: C=1
# 2 prefix: fd2a:0:0:56e0::1/128, ltime: 29min flags: C=1
```

1) Suppose a packet from/to `fd2a:0:0:9a7e::1`: (match = 64)
To my knowledge 6CTX #0 could be used as long as the IID is carried inline

2) Suppose a packet to `fd2a::21f:ff:fe00:758b` (match = 112)
To my knowledge 6CTX #1 could be used:
* 64 must be carried inline if the PANID (in this case 0x1f) of the L2 destination do not match or the 16Bit bit short address do not match
* 16 must be carried inline if the PANID (in this case 0x1f) of the L2 destination match but not the 16 short address
* the address can be elided completely if both PANID and the 16 bit short address of the destination match.

3) Suppose a packet to `fd2a::21f:ff:fe00:758b` (match == 112)
similar compression strategy with (destination replaced by source) as in 2) to applies

However:
`gnrc_sixlowpan_ctx_lookup_addr` return NULL in any of those cases because
_ctx[id].prefixlen (equals 128) < match
(see https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/network_layer/sixlowpan/ctx/gnrc_sixlowpan_ctx.c#L53)

miri64, 2016-11-08:
@melshuber all of the prefixes you configured have bit-length 128. So for the algorithm they are basically all full addresses, so none of the addresses you exemplified match. For your assumptions to be correct you need to setup the following contexts (minimum requirements):

```
# 0 prefix: fd2a::/64, ltime: 29min flags: C=1
# 1 prefix: fd2a::21f:ff:fe00::/112, ltime: 29min flags: C=1
...
```

See https://tools.ietf.org/html/rfc6775#section-5.4.2 for further details on the data structure.

miri64, 2016-11-08:
Note also, that the PANID has no mentioning in https://tools.ietf.org/html/rfc6282. So only IID `00:00:00:ff:fe:00:xx:xx` falls under the rules of compression described there (as is stated there pretty clearly IMHO).

melshuber, 2016-11-09:
@miri64 You have convinced me.
I think the important sentence in the RFC is "**Bits covered by context information are always used.**" (see below). This forces that the complete prefix of the context have to match on order to be to be used

Thanks

https://tools.ietf.org/html/rfc6282#section-3.1.1 :
> [...]
> If SAC=1:
> 
> 00:  The UNSPECIFIED address, ::
> 
> 01: 64 bits.  The address is derived using context information
>  and the 64 bits carried in-line.  **Bits covered by context
>  information are always used.** Any IID bits not covered by
>  context information are taken directly from the corresponding
>  bits carried in-line.  Any remaining bits are zero.
> 
> 10: 16 bits.  The address is derived using context information
>  and the 16 bits carried in-line.  **Bits covered by context
>  information are always used.** Any IID bits not covered by
>  context information are taken directly from their corresponding
>  bits in the 16-bit to IID mapping given by 0000:00ff:fe00:XXXX,
>  where XXXX are the 16 bits carried in- line.  Any remaining bits
>  are zero.
> 
> 11: 0 bits.  The address is fully elided and is derived using
>  context information and the encapsulating header (e.g., 802.15.4
>  or IPv6 source address).  **Bits covered by context information
>  are always used.** Any IID bits not covered by context
>  information are computed from the encapsulating header as
>  specified in Section 3.2.2.  Any remaining bits are zero.

closed 2016-11-09 by melshuber