RIOT-OS/RIOT #6097: Question: memory allocator ?

vincent-d, 2016-11-09:
Hi,

As I am currently working on the CAN stack, I need some mechanisms to allocate memory dynamically.

I used the gnrc_pktbuf_static code, but I had to copy/paste it, as using it as a module would be to intrusive.

I was thinking that maybe such an allocator could be in a separate standalone module. It could also become handy to plug it to the newlib syscalls.

I would like to have a feedback on this, maybe there is already some work on that ?

Thanks

miri64, 2016-11-09:
The first question I would ask is: do you really need it ([money quote](https://twitter.com/Lotterleben/status/754294302503800832) ;-)) or might that be solvable with a static array? The current packet buffer is very constraint to the use case of roughly equal sized datagram traffic.

OlegHahm, 2016-11-09:
First answer: Don't use dynamic memory on embedded devices.
Second answer: Don't misuse packetbuf for that.
Third answer: Don't use dynamic memory. ;-)

If you still want to do it, have you tried using `malloc()`? If `sbrk()` is not correctly implemented, you could use the [TLSF](https://github.com/RIOT-OS/RIOT/tree/master/pkg/tlsf) package (if you're using a 32 bit platform). If you're using a 8 bit or 16 bit platform and don't care about deterministic runtimes, you could also use any other memory allocate (e.g. [umm_malloc](https://github.com/dimonomid/umm_malloc)).

vincent-d, 2016-11-10:
My use case is pretty similar to the gnrc one. I need to allocate buffer for incoming or outcoming packets.

When we stick to raw CAN, the data packets always have the same size, so my first implementation used a  static array which is simpler and probably safer. But when we come to isotp implementation, I need to allocate bigger data packet as an udp implementation could do.

I have three different things I can allocate "dynamically" in my stack:
- raw CAN frames: about 16 bytes (8 data bytes + headers)
- filters object to know which user wants to know that a given frame ID has been received (a few bytes)
- isotp buffer, it can be theorically up to 4096 bytes. The only way of getting rid of dynamic allocation for that use case would be to let the user pass buffers for sending/receiving which makes the user code more complicated and can waste a lot of unused memory.

I would like to avoid using malloc, and would rather keep my memory pool segmented from a general purpose heap.

jcarrano, 2018-06-20:
I'm closing this issue. It is old and we already have TLSF.

closed 2018-06-20 by jcarrano