RIOT-OS/RIOT #6120: Windows AVR Mega development makefile Error

Josar, 2016-11-15:
The RIOT/makefile.include at line 270 inlcudes this
`$(AD)$(OBJCOPY) $(OFLAGS) "$(ELFFILE)" "$(HEXFILE)"`

in for avr-gcc or avr-size the call just prompts the command, for avr-objcopy it inluces the whole path.

This resulte in somthing like that
`/c/Program Files (x86)/Atmel/Studio/7.0/toolchain/avr8/avr8-gnu-toolchain/bin/avr-objcopy -O`

Which does not work and throws an error like 
`/bin/sh: -c: line 0: syntax error near unexpected token `('`

Qoutation marks help in this case.
`$(AD)"$(OBJCOPY)" $(OFLAGS) "$(ELFFILE)" "$(HEXFILE)"`

Kijewski, 2016-11-15:
Please reinstall the toolchain in a path without any spaces. Fixing all problem that spaces introduce is not worth the hassle IMO.

LudwigKnuepfer, 2016-11-15:
I would be interested in whether you have `/c/Program Files (x86)/Atmel/Studio/7.0/toolchain/avr8/avr8-gnu-toolchain/bin/` in your `PATH` or whether you exported `OBJCOPY=/c/Program Files (x86)/Atmel/Studio/7.0/toolchain/avr8/avr8-gnu-toolchain/bin/avr-objcopy`. Not sure how the PATH is handled on Windows.

LudwigKnuepfer, 2016-11-15:
Other than that - I think we did support some space-handling at some point .. has this been removed? If not we should either support it everywhere or remove the partial space support as well.

Josar, 2016-11-16:
To give you some more insight `avr8-gnu-toolchain/bin` is in my PATH. As I mentioned avr-gcc and avr-size are called directly with `avr-gcc`  and `avr-size avr` only objcopy gets expanded with the path. 
So the behaviour here is strange, for my understanding.

I did not export the path and assigned it. What i found is in `cpu/makefile.include.gnu` where an assignment of objcopy is done. 

The withspace support seems only be for filenames, the  ` "$(ELFFILE)"` `"$(HEXFILE)" ` allready had quotation.

Kijewski, 2016-11-16:
> only objcopy gets expanded with the path.

Related line of code: https://github.com/RIOT-OS/RIOT/blob/fe96a4c9c42f94fe0127fba2c3aae450d13f7107/cpu/Makefile.include.gnu#L12

Josar, 2018-01-10:
Does it harm anyhow to but the `"$(OBJCOPY)"` in quatition marks? 

> Qoutation marks help in this case.
$(AD)"$(OBJCOPY)" $(OFLAGS) "$(ELFFILE)" "$(HEXFILE)"

Josar, 2018-01-16:
@kYc0o  as posted in my first post quatation marks for "$(OBJCOPY)" in RIOT/makefile.include at line 270 solves this.

kYc0o, 2018-01-16:
@Josar great! can you make a PR for that? (if you didn't already).

kYc0o, 2018-07-17:
@Josar did you fix this issue?

maribu, 2022-09-16:
Since all Windoofs versions for personal machines that still get security updates do have a Linux Kernel included (called Windoofs Subsystem for Linux or WSL), Windoofs machines now can actually be used for productive work. It still is not the full convenience of a real Linux machine, but it is closer to a sensible setup than Windoofs ever has been.

One remaining issue is that from within WSL it is not possible to access USB devices. (Many Windoofs users claim the opposite, as the file systems of mounted USB flash drives is accessible, but obviously this is not full USB device access.) Windoofs users can work around this by specifying an Windoofs app (often referred to as `.exe` file) as programmer and another Windoofs app a terminal, and `make flash term` should just work.

I think with that, I can close this now. If anyone disagrees, please re-open.

closed 2022-09-16 by maribu