RIOT-OS/RIOT #6123: gnrc: crash with (excessive) traffic

kaspar030, 2016-11-15:
#2071 seems to have fixed native interrupt handling. But I can still crash a riot native instance by flood pinging.

Steps to reproduce:

1. run gnrc_networking
2. start a couple (5-10) ```ping -f <riot-ip>%tap0```
3. wait some minutes

Sometimes it takes 30sec, sometimes 5min, but riot reliably crashes.

The problem seems somewhere in gnrc, as my other stack runs fine since #2071 is merged.
Here's a backtrace:

```
Program received signal SIGSEGV, Segmentation fault.
0x0804df49 in gnrc_pktbuf_start_write ()
(gdb) bt
#0  0x0804df49 in gnrc_pktbuf_start_write ()
#1  0x0805c0ae in _send ()
#2  0x0805ba21 in _event_loop ()
#3  0xf7dba58b in makecontext () from /usr/lib32/libc.so.6
#4  0x00000000 in ?? ()
(gdb)
```

miri64, 2016-11-15:
I have a hunch that this *might* be related to #5748 (or at least fixable with #6086). Can you try with #6086?

miri64, 2016-11-15:
(it might be, that `gnrc_pktbuf_start_write()` needs fixing in #6086 too)

miri64, 2016-11-16:
okay does not seem to be related. Here's a more detailed GDB dump:

```
Program received signal SIGSEGV, Segmentation fault.
0x0804e026 in gnrc_pktbuf_start_write (pkt=0x20) at /home/mlenders/Repositories/RIOT-OS/RIOT/sys/net/gnrc/pktbuf_static/gnrc_pktbuf_static.c:256
256	    if ((pkt == NULL) || (pkt->size == 0)) {
(gdb) where
#0  0x0804e026 in gnrc_pktbuf_start_write (pkt=0x20) at /home/mlenders/Repositories/RIOT-OS/RIOT/sys/net/gnrc/pktbuf_static/gnrc_pktbuf_static.c:256
#1  0x0805c447 in _send (pkt=0x8083ef4 <_pktbuf+756>, prep_hdr=true) at /home/mlenders/Repositories/RIOT-OS/RIOT/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:684
#2  0x0805bd8e in _event_loop (args=0x0) at /home/mlenders/Repositories/RIOT-OS/RIOT/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:279
#3  0xf7e3116b in makecontext () at ../sysdeps/unix/sysv/linux/i386/makecontext.S:91
#4  0x00000000 in ?? ()
```

miri64, 2016-11-16:
Or maybe it is.. sorry about the noise. `pkt=0x20` comes from the packet in `_send`, so this still might be a packet buffer problem...

miri64, 2016-11-16:
(but #6086 does not solve the problem)

kaspar030, 2016-11-16:
> (but #6086 does not solve the problem)

Too bad. Thanks for testing, I didn't get around, yet. So you can reproduce this?

miri64, 2016-11-16:
> So you can reproduce this?

Yepp. But since #6086 doesn't fix #5748 either the problems still seem to be related. But don't have the time atm to look into that. :(

smlng, 2018-01-31:
cannot reproduce with 2018.01-RC2 on macOS native, flood pinged for >20min with 8 instances against 1 RIOT still works

kaspar030, 2018-01-31:
Still happens here, current master, on linux.

kaspar030, 2018-01-31:
> Still happens here, current master, on linux.

8 instances, started manually. after 6-7 first packets get lost and riot prints a couple of "gnrc_netif: possibly lost interrupt.", like every second, then it crashes:

```> gnrc_netif: possibly lost interrupt.
gnrc_netif: possibly lost interrupt.
gnrc_netif: possibly lost interrupt.
gnrc_netif: possibly lost interrupt.
Segmentation fault (core dumped)```

smlng, 2018-01-31:
my test still runs on macOS, RIOT native not crashed after > 1h with 8 parallel ping6 -f on its link-local address. Though, according to Wireshark not all ping request get a reply but many do.

kaspar030, 2018-01-31:
> cannot reproduce with 2018.01-RC2 on macOS native

That might hint that the problem is linux specific. What machine did you run the test on? Maybe this only happens when RIOT doesn't get scheduled enough? My thinkpad is fairly slow by today's standards.

kaspar030, 2018-01-31:
> my test still runs on macOS, RIOT native not crashed after > 1h with 8 parallel ping6 -f on its link-local address.

Can you try increasing to 16 or more?

smlng, 2018-01-31:
latest iMac with i7 @ ~4Ghz and 16GB RAM

> Can you try increasing to 16 or more?

sure!

kaspar030, 2018-01-31:
> latest iMac with i7 @ ~4Ghz and 16GB RAM

That's a 4GHz quad core? Eight ping6 -f are enough to break native on dual 1.6GHz. Do the math, we need to starve the RIOT native thread. ;)

smlng, 2018-01-31:
so 16 parallel pings on RIOT native, still working - but lots of `ping6: sendmsg: No buffer space available" from macOS

smlng, 2018-01-31:
and Wireshark has a hard time to capture all those ping requests and replies ...

bergzand, 2018-01-31:
The other difference between macOS and Linux is that the async_read used by netdev_tap has a different implementation. Maybe that is causing the different observations here.

kYc0o, 2018-07-20:
I tried to reproduce this issue and it really took longtime before crashing, though I didn't exaggerate the number of pings, only 4 instances flooding on vagrant.

Is it expected that RIOT ensures an "eternal" functioning state?

miri64, 2018-07-20:
It is expected that a system exposed to a network is able to either handle the packets or drop them. Crashing is unexceptable as this leaves doors for DoS attacks open. It is unclear though if it is actually GNRC, RIOT, the `native` sublayer or even Linux' TAP infrastructure that is crashing here. Maybe some simple, complexity-increasing tests (first test a TAP application on Linux, than just `netdev_tap`, then `netdev_tap` in a non-main thread, then go to setup explained above) could pin-point the cause of the issue.

miri64, 2018-07-21:
I still can reproduce btw. Maybe in September I find some time to bite myself into this like I did with the leak in `gnrc_pktbuf_static`.

miri64, 2019-01-26:
This issue was re-reported in #10875 with a bit more detailed test description. If the description above is not enough for a tester, please refer to #10875.

miri64, 2019-01-26:
Reproduced it (with different pointers) three more times and one time even with `sudo ping -s0 -f fe80::1c02:edff:fee1:3ec1%tapbr0`, so size doesn't matter here as assessed by @gschorcht in https://github.com/RIOT-OS/RIOT/issues/10875#issuecomment-457830031 is also confirmed. I'll try to reproduce with `gnrc_pktbuf_malloc` and `valgrind` now.

miri64, 2019-01-26:
Was able to reproduce with `valgrind`, but I'm confused by the output.

```
==25483== Invalid read of size 4
==25483==    at 0x120672: _send (sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:533)
==25483==    by 0x11FE55: _event_loop (sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:193)
==25483==    by 0x498927A: makecontext (in /usr/lib32/libc-2.28.so)
==25483==  Address 0x5fd6c98 is 16 bytes inside a block of size 20 free'd
==25483==    at 0x4836857: free (/build/valgrind/src/valgrind/coregrind/m_replacemalloc/vg_replace_malloc.c:530)
==25483==    by 0x1120E5: _release_error_locked (sys/net/gnrc/pktbuf_malloc/gnrc_pktbuf_malloc.c:214)
==25483==    by 0x112165: gnrc_pktbuf_release_error (sys/net/gnrc/pktbuf_malloc/gnrc_pktbuf_malloc.c:228)
==25483==    by 0x12B753: gnrc_pktbuf_release (sys/include/net/gnrc/pktbuf.h:171)
==25483==    by 0x12B7F0: gnrc_pktbuf_remove_snip (sys/net/gnrc/pktbuf/gnrc_pktbuf.c:59)
==25483==    by 0x1206DC: _send (sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:558)
==25483==    by 0x11FE55: _event_loop (sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:193)
==25483==    by 0x498927A: makecontext (in /usr/lib32/libc-2.28.so)
==25483==  Block was alloc'd at
==25483==    at 0x483562B: malloc (/build/valgrind/src/valgrind/coregrind/m_replacemalloc/vg_replace_malloc.c:299)
==25483==    by 0x112301: _create_snip (sys/net/gnrc/pktbuf_malloc/gnrc_pktbuf_malloc.c:276)
==25483==    by 0x111BD8: gnrc_pktbuf_add (sys/net/gnrc/pktbuf_malloc/gnrc_pktbuf_malloc.c:97)
==25483==    by 0x11158D: gnrc_netif_hdr_build (sys/net/gnrc/netif/hdr/gnrc_netif_hdr.c:20)
==25483==    by 0x11E7E9: gnrc_icmpv6_echo_req_handle (sys/net/gnrc/network_layer/icmpv6/echo/gnrc_icmpv6_echo.c:94)
==25483==    by 0x11E252: gnrc_icmpv6_demux (sys/net/gnrc/network_layer/icmpv6/gnrc_icmpv6.c:91)
==25483==    by 0x11FC3F: _demux (sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:112)
==25483==    by 0x120CF5: _receive (sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:799)
==25483==    by 0x11FE42: _event_loop (sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:188)
==25483==    by 0x498927A: makecontext (in /usr/lib32/libc-2.28.so)
```

So it crashes in https://github.com/RIOT-OS/RIOT/blob/bdd2d52fd57c8cc87f9705b2e31c32f58a123a06/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c#L533 because it was already released in https://github.com/RIOT-OS/RIOT/blob/bdd2d52fd57c8cc87f9705b2e31c32f58a123a06/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c#L558 (which comes after...). Either the packet was put into the queue twice or it did not properly dequeue for some reason. Could that be possible @kaspar030?

miri64, 2019-01-26:
Since I am not yet able to reproduce this by adding some `printf`s around the `msg_receive` in `gnrc_ipv6:c:_event_loop`, I'm wondering if this might be related to https://github.com/RIOT-OS/RIOT/issues/1811...

miri64, 2019-01-26:
I was able to reproduce the crash with the following patch:

```diff
diff --git a/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c b/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c
index feb2e8f10..d42696c18 100644
--- a/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c
+++ b/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c
@@ -180,7 +180,11 @@ static void *_event_loop(void *args)
     /* start event loop */
     while (1) {
         DEBUG("ipv6: waiting for incoming message.\n");
+        printf("%d\n", cib_peek((cib_t *)&sched_active_thread->msg_queue));
         msg_receive(&msg);
+        printf("%d (%p)\n", cib_peek((cib_t *)&sched_active_thread->msg_queue),
+               msg.content.ptr);
+
 
         switch (msg.type) {
             case GNRC_NETAPI_MSG_TYPE_RCV:
```

Shortly before the crash I get the following output (using `gnrc_pktbuf_malloc` again, just plane `gdb`:

```
[…]
7
0 (0x565ac0a0)
0
1 (0x565ab1d0)
1
2 (0x565ac060)
2
3 (0x565ab1f0)
3
4 (0x565acd60)
4
5 (0x565ac740)
5
6 (0x565ab190)
6
7 (0x565abfc0)
7
0 (0x565ab210)
0
1 (0x565ac0a0)
1
2 (0x565ab1d0)
2
3 (0x565ac060)
3
4 (0x565ab1f0)
4
5 (0x565acd60)
5
6 (0x565ac740)
6
7 (0x565ab190)
7
-1 (0x565ab230)
0
-1 (0x565a9e50)
-1
-1 (0x565a9e50)

Program received signal SIGSEGV, Segmentation fault.
0x5655efa7 in _release_error_locked (pkt=0x20, err=22) at /home/mlenders/Repositories/RIOT-OS/RIOT/sys/net/gnrc/pktbuf_malloc/gnrc_pktbuf_malloc.c:210
210             tmp = pkt->next;
(gdb) where
#0  0x5655efa7 in _release_error_locked (pkt=0x20, err=22) at /home/mlenders/Repositories/RIOT-OS/RIOT/sys/net/gnrc/pktbuf_malloc/gnrc_pktbuf_malloc.c:210
#1  0x5655f05e in gnrc_pktbuf_release_error (pkt=0x565a9de8, err=22) at /home/mlenders/Repositories/RIOT-OS/RIOT/sys/net/gnrc/pktbuf_malloc/gnrc_pktbuf_malloc.c:228
#2  0x5656d6e9 in _send (pkt=0x565a9de8, prep_hdr=true) at /home/mlenders/Repositories/RIOT-OS/RIOT/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:566
#3  0x5656ce44 in _event_loop (args=0x0) at /home/mlenders/Repositories/RIOT-OS/RIOT/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:197
#4  0xf7d2e27b in makecontext () from /usr/lib32/libc.so.6
#5  0x00000000 in ?? ()
```

Notice the last for lines before the `SIGSEGV`. Weirdly, `msg.content.ptr` also is different from the pointer handed to `_send` as `pkt` (which should be `msg.content.ptr`).

miri64, 2019-01-26:
> ```
> 7
> -1 (0x565ab230)
> 0
> -1 (0x565a9e50)
> -1
> -1 (0x565a9e50)
> ```

I think that can be seen as significant, because these are the first `-1` that comes up once the `ping` starts.

miri64, 2019-01-26:
Also able to reproduce with `gnrc_pktbuf_static`. This time the pointers match.

```
[…]
5
6 (0x5659cdb0)
6
7 (0x5659ce28)
7
0 (0x5659ce40)
0
1 (0x5659d1d0)
1
2 (0x5659cf88)
2
3 (0x5659d1e8)
3
4 (0x5659cf68)
4
5 (0x5659cff8)
5
6 (0x5659d080)
6
7 (0x5659d098)
7
0 (0x5659d118)
0
1 (0x5659ce50)
1
2 (0x5659cd80)
2
3 (0x5659cff8)
3
-1 (0x5659d118)
-1
-1 (0x5659d118)

Program received signal SIGSEGV, Segmentation fault.
0x5656d13f in gnrc_netif_hdr_get_netif (hdr=0x1468) at /home/mlenders/Repositories/RIOT-OS/RIOT/sys/include/net/gnrc/netif/hdr.h:291
291         return gnrc_netif_get_by_pid(hdr->if_pid);
(gdb) where
#0  0x5656d13f in gnrc_netif_hdr_get_netif (hdr=0x1468) at /home/mlenders/Repositories/RIOT-OS/RIOT/sys/include/net/gnrc/netif/hdr.h:291
#1  0x5656ddd4 in _send (pkt=0x5659d118 <_pktbuf+920>, prep_hdr=true) at /home/mlenders/Repositories/RIOT-OS/RIOT/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:543
#2  0x5656d598 in _event_loop (args=0x0) at /home/mlenders/Repositories/RIOT-OS/RIOT/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:197
#3  0xf7d2e27b in makecontext () from /usr/lib32/libc.so.6
#4  0x00000000 in ?? ()
(gdb) call gnrc_pktbuf_stats()
gnrc_netif: possibly lost interrupt.
packet buffer: first byte: 0x5659cd80, last byte: 0x5659e580 (size: 6144)
  position of last byte used: 2984
=========== chunk   0 (0x5659cd80 size:  224) ===========
00000000  B8  CF  59  56  50  CE  59  56  0E  00  00  00  01  00  00  00
00000010  FF  FF  FF  FF  00  00  00  00  00  00  00  00  B0  CE  59  56
[…]
=========== chunk   6 (0x5659d0f0 size:   40) ===========
00000000  60  00  00  00  00  08  3A  40  FE  80  00  00  00  00  00  00
00000010  1C  02  ED  FF  FE  E1  3E  C1  FE  80  00  00  00  00  00  00
00000020  1C  02  ED  FF  FE  E1  3E  C0
~ unused: 0x5659d118 (next: (nil), size: 5224) ~
```

gschorcht, 2019-01-26:
@Good job :smile:

miri64, 2019-01-26:
(btw the IPv6 thread itself is the only one in this scenario sending packets to IPv6)

miri64, 2019-01-26:
Yepp.. definitely no double dispatch:

```
echo:112 0x5659d230
echo:112 0x5659d168
echo:112 0x5659cf28
echo:112 0x5659cf78
echo:112 0x5659d030
echo:112 0x5659d0e0
echo:112 0x5659cef8
echo:112 0x5659d180
echo:112 0x5659d180
echo:112 0x5659d0a0
echo:112 0x5659d198
echo:112 0x5659cf70
echo:112 0x5659d200
echo:112 0x5659cff8
echo:112 0x5659d218
echo:112 0x5659d230
echo:112 0x5659d238
echo:112 0x5659ce88
echo:112 0x5659d078
echo:112 0x5659d0d0
echo:112 0x5659d130
echo:112 0x5659cff0
echo:112 0x5659cfd0
echo:112 0x5659cf80
echo:112 0x5659d1c8
echo:112 0x5659ced0
echo:112 0x5659d1e0
echo:112 0x5659d068
echo:112 0x5659d0c8
echo:112 0x5659d1f8
echo:112 0x5659cf60
echo:112 0x5659d210
echo:112 0x5659d1f8
echo:112 0x5659cfe0
echo:112 0x5659d140
echo:112 0x5659d058
echo:112 0x5659d180
echo:112 0x5659d098
echo:112 0x5659d0b0
echo:112 0x5659cf58
echo:112 0x5659d270
echo:112 0x5659ceb0
echo:112 0x5659d1f0
echo:112 0x5659d0c8
echo:112 0x5659d010
echo:112 0x5659d098
echo:112 0x5659d208
echo:112 0x5659d1b8
echo:112 0x5659d1d0
echo:112 0x5659d1a8
echo:112 0x5659d0d8
echo:112 0x5659ce20
echo:112 0x5659ce98
echo:112 0x5659cf38
echo:112 0x5659cfd0
echo:112 0x5659d048
echo:112 0x5659d0c0
echo:112 0x5659d1d8
echo:112 0x5659d060
echo:112 0x5659d160
echo:112 0x5659ce98
echo:112 0x5659cf38
echo:112 0x5659cf10
echo:112 0x5659d108
echo:112 0x5659d048
echo:112 0x5659d198
echo:112 0x5659d128

Program received signal SIGSEGV, Segmentation fault.
0x5656d0fc in gnrc_netif_hdr_get_netif (hdr=0x18) at /home/mlenders/Repositories/RIOT-OS/RIOT/sys/include/net/gnrc/netif/hdr.h:291
291         return gnrc_netif_get_by_pid(hdr->if_pid);
(gdb) where
#0  0x5656d0fc in gnrc_netif_hdr_get_netif (hdr=0x18) at /home/mlenders/Repositories/RIOT-OS/RIOT/sys/include/net/gnrc/netif/hdr.h:291
#1  0x5656dd38 in _send (pkt=0x5659d128 <_pktbuf+936>, prep_hdr=true) at /home/mlenders/Repositories/RIOT-OS/RIOT/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:539
#2  0x5656d4fc in _event_loop (args=0x0) at /home/mlenders/Repositories/RIOT-OS/RIOT/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:193
#3  0xf7d2e27b in makecontext () from /usr/lib32/libc.so.6
#4  0x00000000 in ?? ()
(gdb)
```

(this was the patch I used)

https://gist.github.com/miri64/a812f7b71fcf6dc71c70d8a9d79b24a9

I'll retry that with the `msg.content.ptr` print from above again, to make 100% sure.

miri64, 2019-01-26:
Yepp no double dispatch...

```diff
diff --git a/sys/net/gnrc/network_layer/icmpv6/echo/gnrc_icmpv6_echo.c b/sys/net/gnrc/network_layer/icmpv6/echo/gnrc_icmpv6_echo.c
index a48ac04f3..0b6f9a73c 100644
--- a/sys/net/gnrc/network_layer/icmpv6/echo/gnrc_icmpv6_echo.c
+++ b/sys/net/gnrc/network_layer/icmpv6/echo/gnrc_icmpv6_echo.c
@@ -109,6 +109,7 @@ void gnrc_icmpv6_echo_req_handle(gnrc_netif_t *netif, ipv6_hdr_t *ipv6_hdr,
 
     LL_PREPEND(pkt, hdr);
 
+    printf("echo:112 %p\n", (void *)pkt);
     if (!gnrc_netapi_dispatch_send(GNRC_NETTYPE_IPV6, GNRC_NETREG_DEMUX_CTX_ALL,
                                    pkt)) {
         DEBUG("icmpv6_echo: no receivers for IPv6 packets\n");
diff --git a/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c b/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c
index feb2e8f10..e5dc0f32f 100644
--- a/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c
+++ b/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c
@@ -181,6 +181,7 @@ static void *_event_loop(void *args)
     while (1) {
         DEBUG("ipv6: waiting for incoming message.\n");
         msg_receive(&msg);
+        printf("ipv6:184 %p\n", msg.content.ptr);
 
         switch (msg.type) {
             case GNRC_NETAPI_MSG_TYPE_RCV:
```

```
ipv6:184 0x5659d0f0
ipv6:184 0x5659cd80
echo:112 0x5659d148
ipv6:184 0x5659ce08
echo:112 0x5659cea8
ipv6:184 0x5659d148
ipv6:184 0x5659cea8
ipv6:184 0x5659cea8
<crash>
```

miri64, 2019-01-26:
@kaspar030 it looks more and more that this is some issue in either `core` or `native`, since the message somehow ends up in `wait_data` I rather suspect the first.

miri64, 2019-01-26:
### Summary

To summarize my findings from above (so you don't have to read all that):

1. A `GNRC_NETAPI_MSG_TYPE_SND` message is received twice in a row by the `gnrc_ipv6` thread *from itself* at some point causing the thread to access parts of the packet released in https://github.com/RIOT-OS/RIOT/blob/bdd2d52fd57c8cc87f9705b2e31c32f58a123a06/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c#L558 
2. The first time the message is retrieved from the thread's message queue. At the second receive the queue is empty and the message comes from `thread_t::wait_data`.
3. `msg_send_to_self()` only queues the packet.
4. It was confirmed that the packet is not dispatched twice for sending by the IPv6 thread, so at the moment the bug seems to be somewhere in the IPC state machine.
5. *Bonus (not found above):* There is no other `msg_send()` that could cause this (confirmed with `git grep msg_send`)

miri64, 2019-01-26:
When playing around with the debugger, I noticed, that `me->wait_data` of `gnrc_ipv6` is always set to the msg` variable of the `_event_loop()` function

closed 2019-02-01 by kaspar030

miri64, 2019-02-01:
:dancer:

PR #10891 (2019-02-01): cpu/native: fix race in thread_yield_higher()
### Contribution description

    Error case:
    1. thread_yield_higher() stores the thread's ucontext
    2. creates an "isr ucontext" for isr_thread_yield, switches to it
    
    Case 1: no signals are pending, continues in isr_thread_yield()
    3a. sched_run is called
    4a. return to sched_active_thread ucontext
    
    Case 2: signals pending (the crashing scenario), continues in native_irq_handler()
    3b. handles signals
    4b. if sched_context_switch_request is set, call sched_run
    5b. return to sched_active_thread ucontext
    
  [... cut]
PR #10921 (2019-02-01): cpu/native: fix race in thread_yield_higher() [backport 2019.01]
# Backport of #10891

### Contribution description

    Error case:
    1. thread_yield_higher() stores the thread's ucontext
    2. creates an "isr ucontext" for isr_thread_yield, switches to it
    
    Case 1: no signals are pending, continues in isr_thread_yield()
    3a. sched_run is called
    4a. return to sched_active_thread ucontext
    
    Case 2: signals pending (the crashing scenario), continues in native_irq_handler()
    3b. handles signals
    4b. if sched_context_switch_request is set, call sched_run
    5b. retu [... cut]