RIOT-OS/RIOT #6125: Question: how to write to a file in native mode

lahmer, 2016-11-16:
HI, I am trying to use riot in simulation in native mode and i want to write all the events to a log file how do i proceed by this ?

miri64, 2016-11-16:
I'm not sure what you mean by "event" and if you just want to log stuff or if you want to log some internal behavior (e.g. ISR triggers, etc). For the latter I would recommend using the `real_open()`, `real_write()`, `real_close()` system calls to the host system in e.g. `cpu/native/irq_cpu.c` to write stuff into your log file. For the first I would suggest to wait for the file system support which is hopefully coming soon-ish to master (see #5616). It hopefully won't be long until we have some shared files with the host system then (if you want to do it hacky you can of course use the system calls mentioned above anywhere else too of course, but I would not recommend merging this to master ;-)).

PeterKietzmann, 2016-11-24:
Have a look at the latest progress in #6072

LudwigKnuepfer, 2016-12-03:
@lahmer if this does not answer your question, please clarify.

OlegHahm, 2016-12-08:
Please shout if your question was not answered.

closed 2016-12-08 by OlegHahm