RIOT-OS/RIOT #6126: nanocoap cannot handle more than 2 endpoints

aabadie, 2016-11-16:
I tried to update some of my CoAP application with the new nanocoap server and I discovered this issue : I have CoAP servers exposing more than 2 custom endpoints, the first 2 can be retrieved but after I get a 4.04 error.

This can be easily reproduced on IoT-LAB with a modified version of the nanocoap_server example :
* Modify the `examples/nanocoap_server/coap_handler.c` file as follows (add 2 extra endpoints, "/mcu" and "/board2"):
```c
[...]
static ssize_t _riot_mcu_handler(coap_pkt_t *pkt, uint8_t *buf, size_t len)
{
    return coap_reply_simple(pkt, buf, len,
            COAP_FORMAT_TEXT, (uint8_t*)RIOT_MCU, strlen(RIOT_MCU));
}

static ssize_t _riot_board2_handler(coap_pkt_t *pkt, uint8_t *buf, size_t len)
{
    return coap_reply_simple(pkt, buf, len,
            COAP_FORMAT_TEXT, (uint8_t*)RIOT_BOARD, strlen(RIOT_BOARD));
}

const coap_resource_t coap_resources[] = {
    COAP_WELL_KNOWN_CORE_DEFAULT_HANDLER,
    { "/riot/board", COAP_GET, _riot_board_handler },
    { "/riot/mcu", COAP_GET, _riot_mcu_handler },
    { "/riot/board2", COAP_GET, _riot_board2_handler },
};

[...]
```
* Follow the [RIOT CoAP tutorial on IoT-LAB](https://www.iot-lab.info/tutorials/coap-riot-with-a8-m3-nodes/)
* Query the different endpoints:
```
$ coap get coap://[IPv6]/riot/board
(2.05)	iotlab-a8-m3
$ coap get coap://[IPv6]/riot/mcu
(2.05)	stm32f1
$ coap get coap://[IPv6]/riot/board2
(4.04)
```
* The `./well-known/core` is correct btw:
```
$ coap get coap://[IPv6]/.well-known/core
(2.05)	</.well-known/core>,</riot/board>,</riot/mcu>,</riot/board2>
```

I tried to find what could be problem but I had a hard time trying to understand the nanocoap internal implementation (adding code comments there would be helpful) and finally gave up.

kaspar030, 2016-11-16:
The endpoints have to be sorted alphabetically...

kaspar030, 2016-11-16:
> The endpoints have to be sorted alphabetically...

... by their path. I'll add some comments.

aabadie, 2016-11-16:
> The endpoints have to be sorted alphabetically...

Thanks ! It works with the endpoints paths sorted !

miri64, 2016-11-16:
Where's the doc-fix?

kaspar030, 2016-11-16:
> Where's the doc-fix?

Not nice changing the issue in mid-flight. If you insist, please also change the issue title.

The doc fix is here: #6132.

miri64, 2016-11-16:
Not nice providing an issue or PR before closing the issue that raised the need for it ;-)

kaspar030, 2016-11-16:
> Not nice providing an issue or PR before closing the issue that raised the need for it ;-)

what are you talking about?

kaspar030, 2016-11-16:
@miri64 Just to clear this up: @aabadie suggested a bug in nanocoap (which happens to be code I wrote), and you assign the issue to me. I identify the "bug" as user error (due to lack of documentation), promise to improve the docs, wait for @aabadie to report back that his stuff is working now and close the issue.
You then re-interpret missing documentation as bug in the documentation, re-tag the issue and re-open it. 
That feels wrong to me on many levels. And please stop throwing emoticons at me.

miri64, 2016-11-16:
Well the doc fix is provided now, so we can close now.

closed 2016-11-16 by miri64

miri64, 2016-11-16:
> @miri64 Just to clear this up: @aabadie suggested a bug in nanocoap (which happens to be code I wrote), and you assign the issue to me. I identify the "bug" as user error (due to lack of [...]
You then re-interpret missing documentation as bug in the documentation, re-tag the issue and re-open it. 
That feels wrong to me on many levels. And please stop throwing emoticons at me.

I just wanted to make sure the doc fix does not get lost. That's all.