RIOT-OS/RIOT #6139: pyterm: prints to stderr per default

miri64, 2016-11-17:
`pyterm` prints to `stderr` per default (and I'm not sure changing this to stdout is configurable). This makes things very difficult if we ever want to use software like [`pexpect` for testing](http://stackoverflow.com/questions/27179383/extracting-stderr-from-pexpect) (the proposed solution on StackOverflow does not allow for input via `child.sendline()` anymore).

OlegHahm, 2016-11-17:
I don't see why pyterm should print to `stderr` and a simple test seems to confirm that this is not the case on my PC. pyterm uses the default root logger for printing on the console, so maybe your Python configuration is somehow different?

miri64, 2016-11-17:
```
[mlenders@sarajevo RIOT.ssh]<3 BOARD=samr21-xpro make -C examples/gnrc_networking/ flash term 2> /dev/null
make: Entering directory '/home/mlenders/Repositories/RIOT-OS/RIOT.ssh/examples/gnrc_networking'
Building application "gnrc_networking" for "samr21-xpro" with MCU "samd21".

"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/boards/samr21-xpro
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/core
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/cpu/samd21
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/cpu/cortexm_common
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/cpu/sam21_common
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/cpu/sam21_common/periph
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/cpu/samd21/periph
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/drivers
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/drivers/at86rf2xx
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/drivers/netdev2_ieee802154
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/drivers/periph_common
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/auto_init
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/auto_init/netif
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/bitfield
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/crosslayer/inet_csum
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/crosslayer/netopt
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/netapi
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/netif
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/netif/hdr
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/netreg
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/network_layer/icmpv6
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/network_layer/icmpv6/echo
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/network_layer/ipv6
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/network_layer/ipv6/hdr
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/network_layer/ipv6/nc
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/network_layer/ipv6/netif
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/network_layer/ndp
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/network_layer/ndp/internal
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/network_layer/sixlowpan
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/network_layer/sixlowpan/ctx
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/network_layer/sixlowpan/frag
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/network_layer/sixlowpan/iphc
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/network_layer/sixlowpan/nd
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/network_layer/sixlowpan/nd/router
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/network_layer/sixlowpan/netif
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/pkt
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/pktbuf_static
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/pktdump
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/routing/rpl
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/transport_layer/udp
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/gnrc/link_layer/netdev2
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/link_layer/ieee802154
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/network_layer/fib
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/network_layer/ipv6/addr
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/network_layer/ipv6/hdr
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/network_layer/sixlowpan
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/net/transport_layer/udp
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/newlib
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/od
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/ps
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/random
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/random/tinymt32
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/shell
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/shell/commands
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/trickle
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/tsrb
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/uart_stdio
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/universal_address
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT.ssh/sys/xtimer
   text	   data	    bss	    dec	    hex	filename
  82132	    592	  20772	 103496	  19448	/home/mlenders/Repositories/RIOT-OS/RIOT.ssh/examples/gnrc_networking/bin/samr21-xpro/gnrc_networking.elf
/home/mlenders/Repositories/RIOT-OS/RIOT.ssh/dist/tools/openocd/openocd.sh flash
### Flashing Target ###
Done flashing
[... cut ...]
And no, I did not change anything in my python config (except that `alias python=python3`, but as you can see: that does not make a difference for the script). Maybe its because its Python 2? Given that, we should enforce `stdout` even more to have equal output on all platforms. There are efforts made to make [python3 the default in Debian-based systems too](https://wiki.ubuntu.com/Python#Python_3_porting) but given that the current LTS of Ubuntu "just" came out shows, that this might take a looong while

miri64, 2016-11-17:
> Maybe its because its Python 2?

Actually, no. Enforcing Python 3 has the same effect.

miri64, 2016-11-17:
Also, from [python doc](https://docs.python.org/3/howto/logging.html#advanced-logging-tutorial):

> By default, no destination is set for any logging messages. You can specify a destination (such as console or file) by using basicConfig() as in the tutorial examples. If you call the functions debug(), info(), warning(), error() and critical(), they will check to see if no destination is set; and if one is not set, they will set a destination of the console (sys.stderr) and a default format for the displayed message before delegating to the root logger to do the actual message output.

OlegHahm, 2016-11-17:
Ah, okay, then I tested incorrectly. I will prepare a patch.

OlegHahm, 2016-11-17:
See #6140.

PR #6140 (2016-11-17): pyterm: log to `stdout` per default
Fixes #6139.

Does anybody see a need to make this configurable?
closed 2016-11-17 by miri64