RIOT-OS/RIOT #6151: Sema dropping messages

lebrush, 2016-11-21:
When a thread uses semaphores (concretely [`sema_wait_timed`](https://github.com/RIOT-OS/RIOT/blob/master/sys/sema/sema.c#L131)) messages are dropped from the message queue. This causes a thread acting on messages to lose events due to the use of a semaphore. Imagine a thread receiving events from [drivers, other threads, ...] and depending on the message it send a message using lwip. Since the current implementation of [lwip](https://github.com/RIOT-OS/RIOT/blob/master/pkg/lwip/contrib/sys_arch.c#L79) relies on the `sema_wait_timed` method, messages are just dropped. This is a kind of non obvious behaviour and difficult to track. I guess it's also a non desired behaviour.

Anyone has dealt with a similar situation already?

One option would be to modify the `sema` API to avoid this case. However, I rather suggest the following change in the `msg` API in order to be more consistent with the handling of messages.

 - Add `msg_receive_type(msg_t *m, uint16_t mask)` and similar `msg_try_receive_type(...)` where the receiving thread can specify a mask for the messages it is interested in. The other messages are not dropped and keep the same position in the queue.

 - Tackle #1749

If you agree, I'd prepare a PR for the first point.

miri64, 2016-11-21:
Adding kaspar due to the change request to `msg`'s API.

miri64, 2016-11-21:
I'd rather prefer to have semaphore implementation with no IPC whatsoever btw.

miri64, 2016-11-21:
(maybe using thread flags? ;-))

lebrush, 2016-11-22:
Good point @miri64, I'll have a look at it :+1:

lebrush, 2016-11-22:
@miri64 please have a look at #6155

lebrush, 2017-01-19:
#6155 got merged

closed 2017-01-19 by lebrush