RIOT-OS/RIOT #6161: You need help to advance support for Teensy 3.x

RndMnkIII, 2016-11-28:
I dont see any advances in Teensy 3.x since I saw the work in progress in supported platforms some months ago. Any technical problem with the freescale cpu?

OlegHahm, 2016-12-08:
@thomaseichinger, do we?

thomaseichinger, 2016-12-08:
I am not actively working on this and haven' been in a while so there was a lot of interface changes in the meanwhile. 
@RndMnkIII Every helping hand is always welcome. In that case it'd be more starting over I guess.

RndMnkIII, 2016-12-23:
Sorry, I've concentrated in another project that requires me to full time

DipSwitch, 2017-02-15:
The only technical problem I have, is that I technically don't have enough spare time to finish what I started ;) But there is light at the end of the tunnel =)

gebart, 2018-04-11:
Closing since Teensy 3.1 and 3.2 support was added by #8210

closed 2018-04-11 by gebart