RIOT-OS/RIOT #6166: base64_encode calculates wrong output size

nikosft, 2016-11-29:
The formula of base64_encode  for calculating base64_out_size is wrong (RIOT/sys/base64/base64.c  lines 62,63). For example for data_in_size = 16 it outputs 26, whereas the correct size is 24.

x3ro, 2016-11-29:
For easier reference here's [a link to the file](https://github.com/RIOT-OS/RIOT/blob/1e8d76870140c2ab76ed3cebb9e5980bcfc77856/sys/base64/base64.c#L62) at the current point in time.

x3ro, 2016-11-29:
This observation appears to be correct, and a bug in our base64 implementation. [This C implementation I've found](http://stackoverflow.com/a/6782480) appears to calculate the correct output size result using the formula

    4 * ((input_length + 2) / 3)

Comparison to the Wikipedia example values: [Wikipedia example values](https://en.wikipedia.org/wiki/Base64#Output_Padding)

| Input Length | Output (correct) | Output (ours) |
|--------------|------------------|---------------|
| 20           | 28               | 29            |
| 19           | 28               | 30            |
| 18           | 24               | 27            |
| 17           | 24               | 25            |

BytesGalore, 2016-11-29:
@nikosft thx for the pointer on this!
@x3ro I changed the used formula to: `((4 * n / 3) + 3) & ~3` which should fix the wrong calculation.

x3ro, 2016-11-29:
@BytesGalore We should definitely add a test for this, though, right? Does fixing this interfere w/ the rest of the algorithm as it stands?

BytesGalore, 2016-11-29:
> We should definitely add a test for this, though, right?

yup will add some tests, mom.

> Does fixing this interfere w/ the rest of the algorithm as it stands?

not I'm aware of.

BytesGalore, 2016-11-29:
@x3ro now uses your proposed formula and I added a small test to check the encoding size determination based on the wikipedia lengths.

PR #6168 (2017-01-11): sys/base64: corrected output size determination
Until now the output size determination calculated a wrong expected output size when encoding.
This PR also adds an additional unittets. 

Fixes #6166
closed 2017-01-11 by PeterKietzmann