RIOT-OS/RIOT #6172: Quick Start Guide link broken...

kaleb-himes, 2016-12-01:
https://github.com/RIOT-OS/RIOT/wiki/Quick-Start-Guide

No Guide, just offers to let you create a new wiki

mali, 2016-12-02:
I think we could make it point here : https://github.com/RIOT-OS/RIOT/wiki/Introduction  ?

kaleb-himes, 2016-12-02:
When I saw that I was hoping it would provide insight on how to go about setting up a new recursive makefile system in an existing library that you wished to port to RIOT OS. I have a library already structured:

<library root>
              | -> <subdirectory 1 w/ source files>
              | -> <subdirectory 2>
                                         |-> <subdirectory 2.1 w/ source files>

I was looking for a quick start guide on porting existing projects to RIOT OS. I already read https://github.com/RIOT-OS/RIOT/wiki/Introduction and it was unfortunately not very helpful.

My current solution was to copy all the source files out of their respective directories withing the library and place them one of the example directories so they would get compiled along with the main.c in the example. Then called the API's necessary for testing from main.c.

I'd prefer to keep my library that is being ported in-tact and add Makefiles throughout the subdirectories but I am unfamiliar with this make system. I'm more used to working with autotools which use include.am files for subdirectories. Do you have a guide for this sort of thing?

kb2ma, 2016-12-02:
Try building the source documentation, `make doc`, in the top-level directory, which is built in the `doc/doxygen/html` directory. From the main doc page, notice especially the Further Information section at the bottom. That should get you started, and then let us know if you have more questions.

kaleb-himes, 2016-12-02:
Will do. Closing this issue for now as it's diverged from the initial topic of "quick start guide link broken". Thanks for your advice!

closed 2016-12-02 by kaleb-himes