RIOT-OS/RIOT #6179: hello-world example on nucleo-f091 uart not working

stream-mab, 2016-12-06:
I've build docker image on osx for building nucleo-f091 hello-world example. Compilation passed and I used following command to burn image to target:
```
export OPENOCD_CONFIG=/x/RIOT/boards/nucleo-f091/dist/openocd.cfg; export HEXFILE=/x/RIOT/examples/hello-world/bin/nucleo-f091/hello-world.hex; ./dist/tools/openocd/openocd.sh flash
```

Start minicom by:
```
minicom -D /dev/cu.usbmodem1423
```

but there is no output (setup is 115200 8N1). Any ideas what to check? Thanks.

kYc0o, 2016-12-07:
Why don't you simply do `make flash term` ?

stream-mab, 2016-12-07:
Developing on osx so build is done on docker and then I call this command to flash image to board.

kYc0o, 2016-12-07:
Oh ok I see... I develop also in OS X and I can do it natively... Do you have any special reason to use docker for this?

You can still compile in docker and use `make term` after flashing, but I think it should also work on minicom... @aabadie is this port using the right UART?

stream-mab, 2016-12-07:
I need to install on osx correct toolchain and then I can do it natively. But this is out of scope of this issue (uart not working) ;)

kYc0o, 2016-12-07:
I asked because maybe some configurations on pyterm are done to make it work, but I don't think so... anyways, can you try to add in `board.h` [this](https://github.com/RIOT-OS/RIOT/blob/master/boards/waspmote-pro/include/board.h#L45) line to modify the UART output? Try to play with the UART numbers until you have the hello-world message on your output.

stream-mab, 2016-12-07:
I tried that already but it doesn't help. Just to be sure that image is correct I enable debug led and led is on so updating of fw in device works fine. Maybe some uart_init is missing in my board?

kYc0o, 2016-12-09:
I just had the time to test and it works correctly:

```console
2016-12-09 12:22:56,782 - INFO # main(): This is RIOT! (Version: 2017.01-devel-166-gccb91-193.55.177-118.saclay.inria.fr)
2016-12-09 12:22:56,788 - INFO # Hello World!
2016-12-09 12:22:56,789 - INFO # You are running RIOT on a(n) nucleo-f091 board.
2016-12-09 12:22:56,792 - INFO # This board features a(n) stm32f0 MCU.
```

stream-mab, 2016-12-09:
Without any change in board code? I tried today on linux machine just run:
```
export BOARD=nucleo-f091; make && make flash
```

Everything seems to be passed without error but no output on /dev/ttyACM0 (1152008N1 on minicom).

kYc0o, 2016-12-13:
Strange... No I didn't change anything, I tested on current master on OS X with a gcc toolchain:

```console
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.2.1 20151202 (release) [ARM/embedded-5-branch revision 231848]
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

My OpenOCD version:
```console
Open On-Chip Debugger 0.10.0-dev-gdfc6658 (2016-02-18-22:16)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
```

I'll try to check it in Linux but I think @aabadie has already done it. Maybe he can confirm first.

kYc0o, 2016-12-20:
Do you have news on this issue? Can we close it?

stream-mab, 2016-12-20:
I find out that somehow virtual port on stlink doesn't work. I used rs232->usb adapter and connect TX and RX to CN3 and I can get output from RIOT. So with this workaround it basically works. Thanks.

OlegHahm, 2016-12-21:
@haukepetersen, I remember that we had similar problems with other nucleo boards in the past. Do you remember if we've ever found the reason behind this?

haukepetersen, 2017-01-09:
No, actually we never found out what caused the problem...

kYc0o, 2017-01-09:
So should we leave this as an open issue? Close as memo?

kYc0o, 2017-01-13:
Justo closed it as memo, reopen if the problem reappears.

closed 2017-01-13 by kYc0o