RIOT-OS/RIOT #6190: stm32/cortex-M: expose clock initialization [RFC]

lebrush, 2016-12-08:
The current implementation of some of the LPM (at least on stm32) re-initializes calls `cpu_init` in order to re-establish the clock system. This might not be the best behavior being only necessary to initialize the clock but not the entire interrupt vectors, etc. (as done by `cpu_init`). 

In the stm platforms this the `cpu_init` usually calls a static method named `clk_init`:

```C
void cpu_init(void)
{
    /* initialize the Cortex-M core */
    cortexm_init();
    /* initialize system clocks */
    clk_init();
}
```

The idea is to make this `clk_init` non static so the lpm calls can use it as well to re-initialize only the clock (extracted from discussion in https://github.com/RIOT-OS/RIOT/pull/6110#discussion_r90754826).

 - Does it apply to other cortex mcu?
 - Does it make sense to fix before #6160, as the later is probably a long term project?
 - Any comments?

Maybe is a good starting point as well to introduce a bit more complex clocking system to allow dynamic changing of the system frequency at run time. This is quick draft without giving it much though. It will probably be way more complex:

```C
typedef struct {
    uint32_t freq_t;
    uint8_t pll_div;
    uint8_t pll_mul;
    uint8_t clk_src;
    uint8_t flash_latency;
    ...
} clock_conf_t;

int clk_init(clock_conf_t conf)
{
    ...
}
```

 - Any opinions on that?

pwillem, 2016-12-08:
It seems that on a stm32f2 it isn't necessary to re-initialize the complete clock. After wake up from stop mode (LPM_SLEEP) I only had to:
- turn on HSE
- turn on PLLs
- configure the system clock switch

kYc0o, 2016-12-09:
I recently faced a similar problem while jumping from different RIOT firmwares in a cortexm device. I just disabled the clock initialisation and it worked, so I agree that some sort of custom clock initialisation would be very useful for various cases.

haukepetersen, 2016-12-15:
+1 for factoring out the clock initialization for the STM32 CPUs! While we are at it, I think it makes also sense to investigate, if some of the code can further be shared for all stm32 CPUs.

I would suggest to go in small steps here:
- factor out the clock init code into a repeatable, non-static function for the stm32 CPUs only
- implement the (hopefully shared) code for the new `pm` interface (-> #6160)
- lets then see which concepts we can further share for all cortex CPUs

closed 2019-09-10 by stale[bot]