RIOT-OS/RIOT #6194: which version of msp430-gcc is recommended?

justinjoy, 2016-12-09:
Hello,

Although I found MSP430-GCC 5.3.0 from TI didn't work for RIOT as it is because some of functions like `getchar` is added unlike previous version of MSP430-GCC, 
I'd like to be sure about recommended compiler version and future plan.

At the present, is msp430-gcc 4.6.x only option for using RIOT?
Have the latest version of gcc, or clang planned to use?

Thanks,
Justin

gebart, 2016-12-09:
I think everyone agrees that requiring gcc 4.6 for one platform while the others are supporting at least up to gcc 5.4 is not optimal, and is holding back the other platforms from using newer features like C11 etc. The problem is that nobody seem to have the time to invest in working on that issue. 
See also the PR on support for msp430 with newlib #4766
If you have some time to spare on this support, feel free to continue on from where that PR has left off.

OlegHahm, 2016-12-12:
I'm planning to dedicate some time to the MSP430 port during my Christmas holidays, but I cannot promise anything.

closed 2019-09-10 by stale[bot]