RIOT-OS/RIOT #6198: drivers: jc42 compatible temperature sensor driver

bergzand, 2016-12-10:
There is a whole slew of simple I^2C temperature sensors compatible with the JC42.4 standard. The standard is meant for temperature sensors on PC memory modules, but they're often for sale as simple modules. Included in this list, but not limited to is:
- MAX6604
- MCP9808
- AT30TS00
- TSE2004

Common features seem to be an interrupt signal for over/under temperature signaling and some hysteris settings.

I'm almost done writing a driver for the MCP9808. If there's any interest in this, I'm willing to write it as a generic jc42 temperature sensor and send a PR. For now I'm focussing on only reading the current temperature and ignore any interrupt signals.

LudwigKnuepfer, 2016-12-10:
There is definitely always an interest in drivers for standard devices!
Don't worry if the whole feature set of something is not met, work that is done is work that is done ;)
Any suggestions for a cheap breakout board for one of these?

bergzand, 2016-12-13:
I'll see to it that I can build a minimal driver implementation then :)

The cheapest I found is the MCP9808 board, probably because there is an open source pcb design for

bergzand, 2017-01-15:
Fixed in #6292

closed 2017-01-15 by bergzand