RIOT-OS/RIOT #6227: cpu/stm32x: unified timer driver

fjmolinas, 2016-12-15:
During the last release a unified version on the timers was built for all STM32x, changing the config structures for a bunch of them put not all board had there periph_config.h changed accordingly. For example the  LimiFrog v1 using the STM32L151RC uC, still implements the old timer config structure witch doesn't work in the new format.

OlegHahm, 2016-12-15:
The title of this PR looks a bit misleading. Could you modify it?

haukepetersen, 2016-12-16:
Hm, i wonder how this slipped through Murdock...

haukepetersen, 2016-12-16:
Hmm, just re-checked: I could not find any STM32 based board that still uses the old timer configuration... The limifrog builds just fine on the current master...

@fjmolinas could you maybe give some more concrete example where you run into trouble?

fjmolinas, 2016-12-18:
If you look at the new timer structure, defined in periph_cpu_common.h this is the format:
 `/**
 * @brief   Timer configuration
 */
typedef struct {
    TIM_TypeDef *dev;       /**< timer device */
    uint32_t max;           /**< maximum value to count to (16/32 bit) */
    uint32_t rcc_mask;      /**< corresponding bit in the RCC register */
    uint8_t bus;            /**< APBx bus the timer is clock from */
    uint8_t irqn;           /**< global IRQ channel */
} timer_conf_t;`

Where as, if you look at the limifrog the timer structure is:

`/**
 * @brief Timer configuration
 * @{
 */
static const timer_conf_t timer_config[] = {
    /* device, RCC bit, IRQ bit */
    {TIM5, 3, TIM5_IRQn},
};`

So even if it builds I would find it weird if it works right since it's missing elements and they aren't in the right order in the struct.

haukepetersen, 2016-12-21:
I see, have only tried to build it without looking at the code... Fixed with #6256

haukepetersen, 2016-12-21:
and thanks for pointing this out!

PeterKietzmann, 2017-01-10:
#6256 was merged, I'll close this issue. @fjmolinas please ping me if you don't see this issue solved

closed 2017-01-10 by PeterKietzmann