RIOT-OS/RIOT #6229: gnrc_rpl: _update_lifetime

Hyungsin, 2016-12-15:
Hi, I have used TinyRPL and ContikiRPL, but now I'm interested in this new OS and currently looking at the RPL implementation on RIOT-OS :)

A question about void _update_lifetime(void) in gnrc_rpl.c:
I think that '**parent->lifetime**' should be decreased in this function as '**inst->dodag.dao_time**' and '**inst->cleanup**' do, but it does not.  Is this a simple bug? or am I wrong?

Hyungsin, 2016-12-15:
Oh sorry, I think I'm wrong. 
parent->lifetime is updated by gnrc_rpl_parent_update() :)
But I don't understand why **parent->lifetime** is updated differently from **dao_time** and **cleanup**. 

Is there any problem if we initialize and update **parent->lifetime** like **dao_time** and **cleanup** without calling **xtimer_now_usec()** in both gnrc_rpl_parent_update() and _update_lifetime()?

BytesGalore, 2016-12-16:
Hi @Hyungsin, 
if I get you right you wonder why we store the `parent->lifetime` while we do not so for DAO routes, i.e. transit options [1].
The point is that parents need to be maintained by RPL to decide if a node remains in a DODAG or needs to find a new parent [2], while DAO routes can safely expire without dropping the node out.
A DAO route has a certain lifetime and a node does not need to keep that route open or track if it is still the case, so we outsource the maintenance to the FIB [3] which takes care of removing the hop.
Contrary a node needs to know when to ask for DIOs if the parents are about to expire, which probably happens quite often compared to opening DAO routes. Of course this assumption does not hold for all usecases.
We could iterate the FIB entries to find if the parent lifetime expired [4]. However, iterating the FIB to search for the parent which we currently look at is more expensive then using its stored lifetime directly.

[1] https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/routing/rpl/gnrc_rpl_control_messages.c#L387
[2] https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/routing/rpl/gnrc_rpl.c#L261
[3] https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/routing/rpl/gnrc_rpl_control_messages.c#L407
[4] https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/routing/rpl/gnrc_rpl_control_messages.c#L745

Hyungsin, 2016-12-17:
Hi @BytesGalore, thanks for a quick response!

Actually, I think my point is different. Sorry for causing confusion :(

If I understand correctly, **_update_lifetime(void)** updates three timers: 1) parent->lifetime [1], 2) inst->cleanup [2], and 3) inst->dodag.dao_time [3]. **parent->lifetime** may be used for checking validity of parent nodes, **inst->cleanup** for checking validity of RPL instances, and **inst->dodag.dao_time** for periodic DAO transmissions. All of them are stored in either gnrc_rpl_parent[x] or gnrc_rpl_instances. So, my point is not whether they are stored or not, but how to update them.

For updating **dao_time** and **cleanup**, you first initialize them to certain values, and decrease them every 'GNRC_RPL_LIFETIME_UPDATE_STEP' in **_update_lifetime()**. There is no need to use xtimer_now for this update procedure.

In contrast, for updating **lifetime**, you first initialize it to **'xtimer_now + certain value'** [4], and do not decrease this value but gain the same effect by calling **xtimer_now** again in **_update_lifetime()**. 
So, my question is if there is a specific reason to use **xtimer_now** only for updating **lifetime**.

[1] https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/routing/rpl/gnrc_rpl.c#L264
[2]  https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/routing/rpl/gnrc_rpl.c#L281
[3]  https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/routing/rpl/gnrc_rpl.c#L290
[4]  https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/routing/rpl/gnrc_rpl_dodag.c#L270

smlng, 2017-01-19:
@cgundogan any comments from your side, as our resident RPL expert 😄 

@Hyungsin: what you said is not quite right, the function `_update_lifetime` decreases the `cleanup` and `dao_time` timers with each call, but it does **not** change the parent lifetime. For the latter it just checks if the timer is close to expiration and will send a DIS or remove the parent accordingly.

However, IMO you are right, that the handling of these timer values is done inconsistently and could (or should) be adjusted to do it the same way for all timers.

Hyungsin, 2017-01-25:
@smlng Yeah, I'm exactly at the same position and just waiting for a comment from @cgundogan  :)

PR #6935 (2017-04-25): rpl: make parent lifetime consistent with other timers
fixes #6229
closed 2017-04-25 by smlng