RIOT-OS/RIOT #6241: examples: iotivity: examples not built by CI

kaspar030, 2016-12-19:
#6096 added some examples one level deeper in the folder hirarchy, so they don't get picked up by the CI.

Hint: they don't all even compile.

kaspar030, 2017-01-13:
The iotivity example makfiles are a mess. I tried building them during murdock tests, and in some cases, the riot checkout ended up only containing tinycbor files, which is a submodule of iotivity. If this doesn't get fixed, I propose removing the package and it's examples for now.

rzr, 2017-01-17:
I can try to help  to support iotivity 

I think it was caused by:

This change 0ed0f4425cffdd84317998efa8fe19eef561c2fa from @kmaloor
Bring RIOT OS port up-to-date with RIOT OS upstream

Any clue ?

I will ask why:
https://gerrit.iotivity.org/gerrit/16209

https://github.com/RIOT-OS/RIOT/pull/6075 might be needed too

kmaloor, 2017-01-17:
Hello,
It appears that the iotivity package is pointing to an old revision of
IoTivity-Constrained, and needs maintenance.
The upstream project (https://github.com/iotivity/iotivity-constrained) that 
I maintain is evolving and was recently updated to use the new xtimer APIs, and as such,
its own examples build and run successfully against the tip of the RIOT-OS tree. 

You may try assigning this issue to @mattiantonini who authored the
RIOT-OS package, as he may be in a good position to address it.

Thanks!

rzr, 2017-01-17:
I have current master building (with WERROR=0)
with patches in:
https://github.com/RIOT-OS/RIOT/pull/6075

You can check on your devices, with this script snippet:

```
make="make WERROR=0 BOARD=arduino-mega2560"

list="\
 examples/iotivity_examples/client \
 examples/iotivity_examples/client_switch \
 examples/iotivity_examples/server \
 examples/iotivity_examples/br_fw \
"
for example in $list ; do $make -C $example ; done
```

rzr, 2017-01-18:
Are there any tasks pending ?

Feel free to ping me if you need help on iotivity support

This change could be merged to avoid a "WERROR FTBFS":
https://github.com/RIOT-OS/RIOT/pull/6403

https://github.com/RIOT-OS/RIOT/pull/6075#event-927115858

emmanuelsearch, 2017-01-18:
I tried `make buildtest` with #6403 after #6075, and with the latest commit of iotivity-constrained (aed3afb26107a1e32c9891b425f8ae2563b19d6e) and all builds fail...

rzr, 2017-01-19:
Yea additional are changed, I started to upstream a couple but it's not over, support from @mattiantonini  is welcome.

Meanwhile you can track efforts at:

https://gerrit.iotivity.org/gerrit/#/q/topic:C99

mattiantonini, 2017-01-19:
Hi All!
I'm investigating about this problem. I have tried to recompile these examples for samr21-xpro (compiler arm-noneaebi-gcc 4.9.3)  and everything goes right without any warning or error.  The problem is for native (gcc 4.9.4)....

P.S. sorry for my delay, but I'm studying for my last exam before graduation

kmaloor, 2017-01-19:
Hello @mattiantonini 

Thank you for checking in on this.

IoTivity-Constrained has changed a bit since Nov, and its RIOT port compiles
and runs successfully against the latest RIOT OS source (i.e. directly, using port/riot/Makefile).

While you’re investigating this, it might be a good idea to revisit a few things in the
RIOT OS pkg and samples on the RIOT OS tree. (can discuss further over e-mail, if required)

1) Patch for multicast:
 1a) Server side: Would it be possible for you to add support for both 
 link-local and realm-local scopes at the same time? If so, you could submit 
 a patch to IoTivity-Constrained (ipadapter.c) and exclude that from the package.

 1b) Client side: You’re modifying the the definition of oc_do_ip_discovery()
 to send requests over only the realm-local scope. I’m not sure if
 this subsumes the link-local scope, and if it doesn’t, then this change
 strikes me as something more relevant to the networking setup for your samples, 
 and not applicable generally. If so, is there another way to handle this?

2) config.h: This has changed a bit, and will be changing some more in the 
next few weeks as I rebalance some things, and are likely to become stable 
from then on. Is there a different way we can handle its exclusion from the package?
I know you’re currently nullifying it with a patch.

3) Warnings as errors: 
I’m busy working on some design enhancements and 
features for making this OCF certifiable, so I’m unable to address
these myself more immediately. 
@rzr  has submitted a couple of patches to clean up warnings, and you’re
free to pitch in as well, when you have the time.
In the meanwhile, it might help to tweak the Makefiles, setting WERROR=0 
before include $(RIOTBASE)/Makefile.include, and in addition
filtering out Wall, Wextra, and pedantic (just like we do in port/riot/Makefile) so
people can build and test the samples.

More long-term, and once we start making formal releases (which should
be soon), I’d be glad to take over the responsibility for keeping the IoTivity RIOT
package and samples in sync.

Thank you!

cc: @rzr @emmanuelsearch @kaspar030 @aabadie

mattiantonini, 2017-01-20:
Hi all,
I have performed the following tests (on the server example) using the latest master of RIOT, IoTivity-Constrained and the patch by @rzr for CBOR(01org/tinycbor#61):

- Native compilation (GCC 4.9.4) fails for Warning on CBOR
- SAMR21-XPRO compilation (arm-none-eabi-gcc 4.9.3) compiles but there is a ram overflow (~1.7 Kb)

@kmaloor regarding your questions:
1) my examples work only in realm-local, but I think it can be easily adapted to deal with also link-local. Anyway, it should be tested
2) I moved the config.h file inside examples folders for easy tweaking
3) In order to add a new module in RIOT every warning/error should be suppressed. What do you suggest? Now, i'm a bit busy for my exam and thesis.

rzr, 2017-01-20:
@mattiantonini   what is the issue on cbor?

Note there are more changes needed for IC, 
I need to take time to check before submitting them upstream, maybe in a couple of weeks.

mattiantonini, 2017-01-20:
@rzr this:

```
In file included from deps/tinycbor/src/cborencoder.c:31:0:
deps/tinycbor/src/cbor.h:166:6: error: ISO C99 doesn’t support unnamed structs/unions [-Werror=pedantic]
     };
      ^
cc1: all warnings being treated as errors
make[2]: *** [deps/tinycbor/src/cborencoder.o] Error 1
make[1]: *** [all] Error 2
make: *** [/path/to/riot/examples/iotivity_examples/server/bin/native/iotivity.a] Error 2
```

kmaloor, 2017-01-23:
@mattiantonini
Regarding your comment: 
> SAMR21-XPRO compilation (arm-none-eabi-gcc 4.9.3) compiles but there is a ram overflow (~1.7 Kb)
the client/server samples in the IoTivity-Constrained tree currently build for SAMR21-xpro using the
Makefile in port/riot.
Client:
   text	   data	    bss	    dec	    hex	filename
  72548	    284	  31120	 103952	  19610	…/iotivity-constrained/port/riot/bin/samr21-xpro/ocf_app.elf
Server:
   text	   data	    bss	    dec	    hex	filename
  72176	    316	  31200	 103692	  1950c	…/iotivity-constrained/port/riot/bin/samr21-xpro/ocf_app.elf
There's one tweak I'm aware of that would lower bss for builds that exclude the security layer. 
I haven't yet added that. That apart, tweaking parameters in config.h for each application would lower 
the RAM utilization. The current values are merely sample picks. I guess we need to lay out
some of these details more clearly in IoTivity-Constrained's README.

Regarding the other points from my earlier note:
1. Yes, it would be preferable not to patch the client API to send on realm-local. On the receive side however, it should be OK to listen for both link-local and realm-local.
2. Yes, I understand why you put config.h in the folders of each example, and that is the right thing
to do. I guess I was wondering if we could somehow avoid the nullifying patch in the package that removes port/riot/config.h. But may be it is OK if we can't.
3. No worries. I will take care of eradicating all of the build warnings. 

So, lets do this: I will report back here once I've finished with 3. Then you can fix 1. and update the
package.

cc: @rzr

rzr, 2017-03-07:
Note https://github.com/01org/tinycbor/commits/v0.4.1 has previous RIOT fixes I upstreamed , it may worth to check again.

kaspar030, 2017-08-29:
I'll close as the examples have been removed.

closed 2017-08-29 by kaspar030