RIOT-OS/RIOT #6244: Vagrant VM not connecting via SSH

kennethjiang, 2016-12-20:
Vagrant: 1.8.6
Host: Mac OSX 10.11.6
Hypervisor: VirtualBox 5.0.28

`vagrant up` ended with the error message:
```
ERROR vagrant: Vagrant experienced an error! Details:
ERROR vagrant: #<Vagrant::Errors::VMBootTimeout: Timed out while waiting for the machine to boot. This means that
Vagrant was unable to communicate with the guest machine within
the configured ("config.vm.boot_timeout" value) time period.
```

When I turned on Vagrant debug, it timed out at this command:
```
 INFO ssh: Invoking SSH: ssh ["vagrant@127.0.0.1", "-p", "2222", "-o", "Compression=yes", "-o", "DSAAuthentication=yes", "-o", "LogLevel=FATAL", "-o", "IdentitiesOnly=yes", "-o", "StrictHostKeyChecking=no", "-o", "UserKnownHostsFile=/dev/null", "-i", "/Users/kenneth/.vagrant.d/insecure_private_key"]
```

In VirtualBox admin console, the VM was actually running, with login screen on console.

--------
Additional debugging info:

The openssh server running on the guest VM seems to have problems:

```bash
$ nc -v localhost 2222                                                                                      ⏎
nc: connectx to localhost port 2222 (tcp) failed: Connection refused
found 0 associations
found 1 connections:
     1:	flags=82<CONNECTED,PREFERRED>
	outif lo0
	src 127.0.0.1 port 62070
	dst 127.0.0.1 port 2222
	rank info not available
	TCP aux info available

Connection to localhost port 2222 [tcp/ats] succeeded!
```

Whereas a correctly-behaving openssh server would return this:
```bash
nc -v localhost 2222
nc: connectx to localhost port 2222 (tcp) failed: Connection refused
found 0 associations
found 1 connections:
     1:	flags=82<CONNECTED,PREFERRED>
	outif lo0
	src 127.0.0.1 port 62232
	dst 127.0.0.1 port 2222
	rank info not available
	TCP aux info available

Connection to localhost port 2222 [tcp/rockwell-csp2] succeeded!
SSH-2.0-OpenSSH_7.2p2 Ubuntu-4ubuntu2.1
```
Please notice openssh server in RIOT-VM is missing `SSH-2.0-OpenSSH_7.2p2 Ubuntu-4ubuntu2.1` in response.

smlng, 2016-12-20:
@kennethjiang I just tested on macOS 10.11.6 (same as yours) with latest VirtualBox (5.1.10) and Vagrant (1.9.1) and it worked for me. Output of `vagrant up`:

```
ringing machine 'default' up with 'virtualbox' provider...
==> default: Box 'RIOT/ubuntu1604' could not be found. Attempting to find and install...
    default: Box Provider: virtualbox
    default: Box Version: >= 0
==> default: Loading metadata for box 'RIOT/ubuntu1604'
    default: URL: https://atlas.hashicorp.com/RIOT/ubuntu1604
==> default: Adding box 'RIOT/ubuntu1604' (v0.0.1) for provider: virtualbox
    default: Downloading: https://atlas.hashicorp.com/RIOT/boxes/ubuntu1604/versions/0.0.1/providers/virtualbox.box
==> default: Successfully added box 'RIOT/ubuntu1604' (v0.0.1) for 'virtualbox'!
==> default: Importing base box 'RIOT/ubuntu1604'...
==> default: Matching MAC address for NAT networking...
==> default: Checking if box 'RIOT/ubuntu1604' is up to date...
==> default: Setting the name of the VM: RIOT-VM
==> default: Clearing any previously set network interfaces...
==> default: Preparing network interfaces based on configuration...
    default: Adapter 1: nat
==> default: Forwarding ports...
    default: 22 (guest) => 2222 (host) (adapter 1)
==> default: Booting VM...
==> default: Waiting for machine to boot. This may take a few minutes...
    default: SSH address: 127.0.0.1:2222
    default: SSH username: vagrant
    default: SSH auth method: private key
    default: Warning: Remote connection disconnect. Retrying...
    default: 
    default: Vagrant insecure key detected. Vagrant will automatically replace
    default: this with a newly generated keypair for better security.
    default: 
    default: Inserting generated public key within guest...
    default: Removing insecure key from the guest if it's present...
    default: Key inserted! Disconnecting and reconnecting using new SSH key...
==> default: Machine booted and ready!
==> default: Checking for guest additions in VM...
==> default: Mounting shared folders...
    default: /vagrant => /tmp/RIOT
    default: /home/vagrant/RIOT => /tmp/RIOT
==> default: Running provisioner: file...
```
As you see there was a timeout during boot, too, but still worked. Afterwards I called `vagrant ssh` and that was the output
```
Welcome to Ubuntu 16.04.1 LTS (GNU/Linux 4.4.0-36-generic x86_64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage

0 packages can be updated.
0 updates are security updates.

vagrant@vagrant:~$ ls
RIOT  stlink-master
```

Do you installed vbox extension pack? Maybe you system has high load or memory usage, so that vagrant RIOT VM does not get enough resources to run?

kennethjiang, 2016-12-20:
Upgraded VirtualBox to 5.1.10 and it worked! Did not have to upgrade Vagrant.

Thanks @smlng for your prompt response!

closed 2016-12-20 by kennethjiang