RIOT-OS/RIOT #6248: shell/sc_gnrc_6ctx strtok_r missing

BytesGalore, 2016-12-20:
building `examples/gnrc_border_router` throws out the following error when building:
```
...
"make" -C ./RIOT/sys/shell/commands
./RIOT/sys/shell/commands/sc_gnrc_6ctx.c: In function '_gnrc_6ctx_add':
./RIOT/sys/shell/commands/sc_gnrc_6ctx.c:86:16: error: implicit declaration of function 'strtok_r' [-Werror=implicit-function-declaration]
     addr_str = strtok_r(prefix_str, "/", &save_ptr);
                ^
...
```
tested on Ubuntu 16.04.1 LTS with:
`arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.3.1 20160307 (release) [ARM/embedded-5-branch revision 234589]`
`andarm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]`

Building the example with a ~3 weeks old RIOT works flawlessly, so probably something has changed in the newlib?

smlng, 2016-12-20:
I ran `git bisect`, turns out #5824 by @kaspar030 introduces this behavior

gebart, 2016-12-20:
Likely it is similar to an issue I had in #5616 where we need to define
_DEFAULT_SOURCE before including the header. (the issue I had was with
strnlen)

BytesGalore, 2016-12-20:
@gebart your proposition fixes the issue described here :)

kaspar030, 2016-12-20:
I wonder how this slips by murdock.

smlng, 2016-12-20:
@kaspar030 its seems only to happen on arm cortex platforms and compiler arm-none-eabi-gcc version > 5.2. Maybe Murdock has an older version running?

gebart, 2016-12-20:
My guess is that it depends on tool chain versions etc.

smlng, 2016-12-20:
I tested with 5.2 and it worked, but with 5.4 (2016q3) it fails.

closed 2016-12-21 by smlng

kYc0o, 2017-03-07:
Well... It seems that #6250 didn't fix this for older compilers.
I'm on a Debian Jessie with `arm-none-eabi-gcc`"

```console
arm-none-eabi-gcc (4.8.4-1+11-1) 4.8.4 20141219 (release)
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

The error seems to be quite the same, while building examples/gnrc_border_router:

```console
RIOT/sys/shell/commands/sc_gnrc_6ctx.c:88:5: error: implicit declaration of function 'strtok_r' [-Werror=implicit-function-declaration]
     addr_str = strtok_r(prefix_str, "/", &save_ptr);
     ^
RIOT/sys/shell/commands/sc_gnrc_6ctx.c:88:14: error: assignment makes pointer from integer without a cast [-Werror]
     addr_str = strtok_r(prefix_str, "/", &save_ptr);
              ^
RIOT/sys/shell/commands/sc_gnrc_6ctx.c:93:20: error: assignment makes pointer from integer without a cast [-Werror]
     prefix_len_str = strtok_r(NULL, "/", &save_ptr);
                    ^
cc1: all warnings being treated as errors
```

PR #6250 (2016-12-21): shell: fix strtok_r compile error
fixes  #6248