RIOT-OS/RIOT #6261: SAMD21 TC3 & TC4 issue when using TCC2

spiderkeys, 2016-12-22:
Ran into a bug in the pwm driver tonight that led to a bunch of odd observations; things working when they shouldn't be and the like. I'll break down everything the best I can.

**What I did:**
Created a new board that uses the SAMD21 CPU. Configured PWM0 to use TCC2. When pwm_init is called for PWM0, xtimer starts exhibiting very strange behavior and reports time elapsing much faster than it ought to. I commented out code in pwm_init until I narrowed it down to the GCLK configuration step in pwm_poweron(). I noticed here that it was enabling the clock ID for TCC0/TCC1 or TCC2/TC3 on GCLK0, whereas TC3, TC4, and TC5 are configured to use GCLK1 if you are running on the PLL. Added the following code and everything started working normally again:

```c++
    #if CLOCK_USE_PLL
        GCLK->CLKCTRL.reg = (GCLK_CLKCTRL_CLKEN |
                            GCLK_CLKCTRL_GEN_GCLK1 |
                            GCLK_CLKCTRL_ID(_clk_id(dev)));
    #else
        GCLK->CLKCTRL.reg = (GCLK_CLKCTRL_CLKEN |
                            GCLK_CLKCTRL_GEN_GCLK0 |
                            GCLK_CLKCTRL_ID(_clk_id(dev)));
    #endif
    
    while (GCLK->STATUS.bit.SYNCBUSY) {}
```

**Observations of thing working that maybe shouldn't:**
1. I have another custom board which is identical in all ways except that it doesn't use TCC2. PWM somehow works without my above code which checks CLOCK_USE_PLL, despite the fact that the TCC0/TCC1 clocks are being enabled on GCLK0 instead of GCLK1.
2. To my knowledge, TC4 is not in any way tied to TCC2. I know TC3 and TCC2 share the same clock ID, so I changed my XTIMER to point to TIMER_1 which is defined to use TC4. TC4 is configured by timer_init() to use GCLK1 as it is configured to use the PLL. Configuring the clock for TCC2 should not have affected TC4, and thus should not have affected xtimer_now() results in any way.

Any ideas?

Some additional details concerning my setup:

**Hardware:**
SAMD21J18A using an external oscillator to drive the PLL and give me a 48MHz clock (same clock configuration as the Arduino Zero board and Sodaq Autonomo board).

**PWM configuration:**
```c++
// PWM Configuration
#define PWM_0_EN            1
#define PWM_1_EN            1
#define PWM_MAX_CHANNELS    4

/* for compatibility with test application */
#define PWM_0_CHANNELS      PWM_MAX_CHANNELS
#define PWM_1_CHANNELS      PWM_MAX_CHANNELS

/* PWM device configuration */
static const pwm_conf_t pwm_config[] = 
{
#if PWM_0_EN
    {   TCC2, 
        {
            /* GPIO pin, MUX value, TCC channel */
            { GPIO_PIN(PA, 16), GPIO_MUX_E, 0 }, // Lights
            { GPIO_UNDEF, (gpio_mux_t)0, 1 },
            { GPIO_UNDEF, (gpio_mux_t)0, 2 },
            { GPIO_UNDEF, (gpio_mux_t)0, 3 }
        }
    }
#endif
#if PWM_1_EN
    ,{  TCC0, 
        {
            /* GPIO pin, MUX value, TCC channel */
            { GPIO_PIN(PB, 10), GPIO_MUX_F, 0 },    // MotorA
            { GPIO_PIN(PB, 11), GPIO_MUX_F, 1 },    // MotorB
            { GPIO_PIN(PB, 12), GPIO_MUX_F, 2 },    // MotorC
            { GPIO_UNDEF, (gpio_mux_t)0, 3 }     
        }
    }
#endif
};

// PWM Device count
#define PWM_NUMOF           (2U)
```

**Timer Config:**
```c++
#define XTIMER              TIMER_1
#define XTIMER_CHAN         (0)
```
```c++
// Configure Timer Peripherals
#define TIMER_NUMOF         (2U)
#define TIMER_0_EN          1
#define TIMER_1_EN          1

// Timer 0 configuration
#define TIMER_0_DEV         TC3->COUNT16
#define TIMER_0_CHANNELS    2
#define TIMER_0_MAX_VALUE   (0xffff)
#define TIMER_0_ISR         isr_tc3

// Timer 1 configuration
#define TIMER_1_DEV         TC4->COUNT32
#define TIMER_1_CHANNELS    2
#define TIMER_1_MAX_VALUE   (0xffffffff)
#define TIMER_1_ISR         isr_tc4
```

spiderkeys, 2016-12-22:
For full application source see:
https://gitlab.com/openrov/RIOT/tree/debug/rev8

Application directory:
apps/trident

BOARD=openrov-trident-rev8

Note: I have made some other bug fixes to the PWM driver that I haven't submitted back to RIOT yet, so there may be discrepancies between my source and master.

travisgriggs, 2017-04-18:
Is there a PR for this? Or did it stall out? We'd really like to be able to use the clock configuration mentioned (it's the best configuration for really low power consumption).

PeterKietzmann, 2017-11-27:
@immesys I know you are really experienced with "low power" (in general) on that platform. Could you share some insights?

photonthunder, 2017-11-27:
@PeterKietzmann the latest version of RIOT does now include an xosc32/DFLL option.  Discussions are ongoing about coming up with low power options (issue #7332, #7743, #7913, #7944).  That being said, any insight that @immesys or anyone else provides to get the ball rolling would be greatly appreciated.

immesys, 2017-11-27:
@hyungsin want to look over this? I'll take a look later

PeterKietzmann, 2018-06-26:
@spiderkeys, @photonthunder, @immesys anyone still there to report whether this is still an open issue?

spiderkeys, 2018-06-26:
@PeterKietzmann Sorry, we moved away from PWM in our application (this was for talking to a motor controller and we now use UART), so I can no longer comment on whether or not this is still an issue.

photonthunder, 2018-06-26:
@PeterKietzmann the basic functionality is now in the RIOT code (to use DFLL instead of PLL).  Attempts to "clean up" further have stalled out (see PR #7315 and issue #7913).

PeterKietzmann, 2018-06-26:
Thanks for your feedback. It seems most points have been addressed though it's not fully clear if the initial problem persists. I'll close with memo label set. If you have the impression this is the wrong way to go, please give me a sign.

closed 2018-06-26 by PeterKietzmann