RIOT-OS/RIOT #6264: arduino uno - threads doesn't work anymore

mali, 2016-12-22:
tried with ipc_pingpong example, 2nd thread don't start.
I'm sure it was working correctly,
I'll try to git bisect later if nobody doesn't come with a solution before.

mali, 2016-12-22:
Hum ... git bisect doesn't help

OlegHahm, 2016-12-22:
What do you mean? Why did `git bisect` not help? Maybe a toolchain update?

mali, 2016-12-27:
I'd like someone else to try to build and run ipc_pingpong example and say if it work's ...
I'd suspect my setup is broken but can't figure why ...

kYc0o, 2016-12-27:
I can confirm that it doesn't work in arduino-uno... I'll try to check why but not very soon...

kYc0o, 2016-12-27:
However it does work for arduino-mega2560... And it's because there's no definition for `AVR_CONTEXT_SWAP_INIT`. Look [here](https://github.com/RIOT-OS/RIOT/blob/master/boards/arduino-mega2560/include/board.h#L56) for an example. Can you maybe provide a PR to fix it?

mali, 2016-12-27:
AVR_CONTEXT_SWAP_INIT is defined [here](https://github.com/RIOT-OS/RIOT/blob/master/boards/arduino-common/include/board.h#L55) in arduino-common, but I'll dig deeper in this direction

kYc0o, 2016-12-27:
Yes I just saw it, but I don't know if it was tested before to make sure that it works. Also I can see a confusing thing about "arduino-common" since there are several other arduino boards which don't use this files... anyways, maybe rename them is another issue, I'll try to not forget it and find a consensus on this.

mali, 2016-12-27:
Yes, it was working, and it should trigger an [error](https://github.com/RIOT-OS/RIOT/blob/master/cpu/atmega_common/thread_arch.c#L31) if undefined.
arduino-common is in fact common to arduino uno and duemilanove, maybe arduino-atmega328p-common would be a better name ....

A-Paul, 2017-01-09:
I recently started to adapt to a "arduino pro mini" clone (also atmega328p) and the shell isn't working. That is related to this bug, I think?

kYc0o, 2017-01-12:
Solved by #6320

closed 2017-01-12 by kYc0o