RIOT-OS/RIOT #6267: Label cleanu

x3ro, 2016-12-23:


LudwigKnuepfer, 2016-12-25:
I added and applied the label (and removed it) only for illustration purposes.

LudwigKnuepfer, 2016-12-25:
Initially I liked the idea.

However, looking at https://github.com/himynameisdave/git-labelmaker/issues?q=is%3Aissue+is%3Aclosed I have the impression that it gets more difficult to actually read the labels because every second word in the plaintext is metadata.
Also this type of metadata is kind of redundant as the titles themselves already imply a type.
Tool-wise it does not make a difference if it searches for "Status: Ready for CI Build" or "Ready for CI Build".

All in all I conclude that it would be nice to have metadata for the purpose of browsing issues, but that integration into Github is suboptimal and that we don't gain anything else.

LudwigKnuepfer, 2016-12-25:
I would like to see what the JSON label file looks like and how it integrates, but that is a separate issue IMHO.

x3ro, 2017-11-28:
Closing this, since it didn't generate any visible interest. Feel free to bring it up again if you're interested.

closed 2017-11-28 by x3ro

miri64, 2017-11-28:
@danpetry maybe this is something you can pick up?

danpetry, 2017-11-29:
Hi @miri64 , yep this is something I'm interested in, I have made a note and will re-open when I have time to pay it more attention.

danpetry, 2018-07-04:
Some examples of good labelling:
https://github.com/ARMmbed/mbed-os/pulls
https://github.com/rtrlib/rtrlib/pulls
https://github.com/rust-lang/rust/pulls
In general prefixes look like a good way to go...

miri64, 2018-07-04:
See also #9449.

miri64, 2018-09-25:
ANd #10030