RIOT-OS/RIOT #6279: GNRC doesn't do stateless address autoconfiguration based on router advertisement

kennethjiang, 2017-01-02:
Board: samr21-xpro
Branch: master

Steps to reproduce:
1. Install `radvd` on Raspberry Pi, and configure it to advertise on `lowpan0` network interface.
2. Build microcoap server: `cd RIOT/examples/microcoap_server && make all flash BOARD=samr21-xpro`
3. `make term BOARD=samr21-xpro` and watch the output of `_netif_config`.

Bug: Tcpdump shows that radvd responds to router solicitation with correct advertisement, including prefix information option (included below). However, GNRC fails to configure network interface with appropriate IPv6 address and routing.

Tcpdump:
```
21:45:14.391751 IP6 (hlim 255, next-header ICMPv6 (58) payload length: 24) fe80::5855:4443:9918:b576 > ff02::2: [icmp6 sum ok] ICMP6, router solicitation, length 24
	  source link-address option (1), length 16 (2): 5a:55:44:43:99:18:b5:76:00:00:00:00:00:00
	    0x0000:  5a55 4443 9918 b576 0000 0000 0000
	0x0000:  6000 0000 0018 3aff fe80 0000 0000 0000  `.....:.........
	0x0010:  5855 4443 9918 b576 ff02 0000 0000 0000  XUDC...v........
	0x0020:  0000 0000 0000 0002 8500 a3d5 0000 0000  ................
	0x0030:  0102 5a55 4443 9918 b576 0000 0000 0000  ..ZUDC...v......
21:45:14.392912 IP6 (hlim 255, next-header ICMPv6 (58) payload length: 48) fe80::effe:44df:9b85:ef7a > ff02::1: [icmp6 sum ok] ICMP6, router advertisement, length 48
	hop limit 64, Flags [none], pref low, router lifetime 30s, reachable time 0s, retrans time 0s
	  prefix info option (3), length 32 (4): fd2d:388:6a7b:2::/120, Flags [onlink, auto], valid time 86400s, pref. time 14400s
	    0x0000:  78c0 0001 5180 0000 3840 0000 0000 fd2d
	    0x0010:  0388 6a7b 0002 0000 0000 0000 0000
	0x0000:  6000 0000 0030 3aff fe80 0000 0000 0000  `....0:.........
	0x0010:  effe 44df 9b85 ef7a ff02 0000 0000 0000  ..D....z........
	0x0020:  0000 0000 0000 0001 8600 0b42 4018 001e  ...........B@...
	0x0030:  0000 0000 0000 0000 0304 78c0 0001 5180  ..........x...Q.
	0x0040:  0000 3840 0000 0000 fd2d 0388 6a7b 0002  ..8@.....-..j{..
	0x0050:  0000 0000 0000 0000                      ........
```

Samr21 term output (ENABLE_DEBUG=1):
```
2017-01-02 02:25:42,088 - INFO # ipv6 netif: Get IPv6 interface 6 (0x20001660, i = 0)
2017-01-02 02:25:42,092 - INFO # ipv6 netif: Found ff02::1 on interface 6
2017-01-02 02:25:42,095 - INFO # icmpv6: router advertisement received
2017-01-02 02:25:42,100 - INFO # ipv6 netif: Get IPv6 interface 6 (0x20001660, i = 0)
2017-01-02 02:25:42,312 - INFO # ipv6 netif: Get IPv6 interface 6 (0x20001660, i = 0)
2017-01-02 02:25:42,318 - INFO # ndp: Router advertisement did not contain any source address information
2017-01-02 02:25:42,323 - INFO # ipv6 netif: Get IPv6 interface 6 (0x20001660, i = 0)
2017-01-02 02:25:42,327 - INFO # ipv6 netif: Get IPv6 interface 6 (0x20001660, i = 0)
2017-01-02 02:25:42,338 - INFO # ndp internal: send neighbor solicitation (iface: 6, src: ::,  tgt: fe80::effe:44df:9b85:ef7a, dst: fe80::effe:44df:9b85:ef7a)
2017-01-02 02:25:42,343 - INFO # ipv6 netif: Get IPv6 interface 6 (0x20001660, i = 0)
2017-01-02 02:25:42,345 - INFO # gathering candidates
2017-01-02 02:25:42,347 - INFO # Checking address: ff02::1
2017-01-02 02:25:42,351 - INFO # Checking address: fe80::5855:4443:9918:b576
2017-01-02 02:25:42,353 - INFO # add to candidate set
2017-01-02 02:25:42,356 - INFO # Checking address: ff02::1:ff18:b576
2017-01-02 02:25:42,358 - INFO # Checking address: ff02::2
2017-01-02 02:25:42,360 - INFO # Checking address: ::
2017-01-02 02:25:42,362 - INFO # Checking address: ::
2017-01-02 02:25:42,364 - INFO # Checking address: ::
2017-01-02 02:25:42,366 - INFO # Checking address: ::
2017-01-02 02:25:42,371 - INFO # finding the best match within the source address candidates
2017-01-02 02:25:42,373 - INFO # Checking address: ff02::1
2017-01-02 02:25:42,377 - INFO # Not part of the candidate set - skipping
2017-01-02 02:25:42,381 - INFO # Checking address: fe80::5855:4443:9918:b576
2017-01-02 02:25:42,384 - INFO # winner for rule 2 (same scope) found
2017-01-02 02:25:42,387 - INFO # Checking address: ff02::1:ff18:b576
2017-01-02 02:25:42,390 - INFO # Not part of the candidate set - skipping
2017-01-02 02:25:42,393 - INFO # Checking address: ff02::2
2017-01-02 02:25:42,396 - INFO # Not part of the candidate set - skipping
2017-01-02 02:25:42,400 - INFO # Checking address: ::
2017-01-02 02:25:42,403 - INFO # Not part of the candidate set - skipping
2017-01-02 02:25:42,404 - INFO # Checking address: ::
2017-01-02 02:25:42,407 - INFO # Not part of the candidate set - skipping
2017-01-02 02:25:42,409 - INFO # Checking address: ::
2017-01-02 02:25:42,412 - INFO # Not part of the candidate set - skipping
2017-01-02 02:25:42,414 - INFO # Checking address: ::
2017-01-02 02:25:42,418 - INFO # Not part of the candidate set - skipping
2017-01-02 02:25:42,430 - INFO # ipv6 netif: Found fe80::5855:4443:9918:b576 on interface 6 matching fe80::effe:44df:9b85:ef7a by hu bits (used as source address = =)
2017-01-02 02:25:42,437 - INFO # ndp: building source link-layer address option (l2addr: 5a:55:44:43:99:18:b5:76)
2017-01-02 02:25:42,441 - INFO # ndp: building neighbor solicitation message
2017-01-02 02:25:42,445 - INFO # icmpv6: Building ICMPv6 message with type=135, code=0
2017-01-02 02:25:42,450 - INFO # ipv6 netif: Get IPv6 interface 6 (0x20001660, i = 0)
2017-01-02 02:25:42,453 - INFO # icmpv6: no one interested in type 134
2017-01-02 02:25:42,458 - INFO # ipv6 netif: Get IPv6 interface 6 (0x20001660, i = 0)
2017-01-02 02:25:42,463 - INFO # ipv6 netif: Get IPv6 interface 6 (0x20001660, i = 0)
2017-01-02 02:25:42,468 - INFO # ipv6 netif: Get IPv6 interface 6 (0x20001660, i = 0)
2017-01-02 02:25:42,479 - INFO # ipv6 netif: Get IPv6 interface 6 (0x20001660, i = 0)
2017-01-02 02:25:42,484 - INFO # ipv6 netif: Found fe80::5855:4443:9918:b576 on interface 6
2017-01-02 02:25:42,488 - INFO # icmpv6: neighbor advertisement received
2017-01-02 02:25:42,500 - INFO # ndp: received neighbor advertisement (src: fe80::effe:44df:9b85:ef7a, dst: fe80::5855:4443:9918:b576, tgt: fe80::effe:44df:9b85:ef7a)
2017-01-02 02:25:42,509 - INFO # ndp internal: set fe80::effe:44df:9b85:ef7a state to ipv6 netif: Get IPv6 interface 6 (0x20001660, i = 0)
2017-01-02 02:25:42,513 - INFO # REACHABLE (reachable time = 18000000 us)
2017-01-02 02:25:42,516 - INFO # icmpv6: no one interested in type 136
2017-01-02 02:25:43,688 - INFO # Configured network interfaces:
2017-01-02 02:25:43,693 - INFO # ipv6 netif: Get IPv6 interface 6 (0x20001660, i = 0)
2017-01-02 02:25:43,698 - INFO # Iface  6   HWaddr: 35:76  Channel: 26  Page: 0  NID: 0x23
2017-01-02 02:25:43,703 - INFO #            Long HWaddr: 5a:55:44:43:99:18:b5:76
2017-01-02 02:25:43,709 - INFO #            TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4
2017-01-02 02:25:43,716 - INFO #            ACK_REQ  CSMA  MTU:1280  HL:64  6LO  RTR  RTR_ADV  IPHC
2017-01-02 02:25:43,719 - INFO #            Source address length: 8
2017-01-02 02:25:43,722 - INFO #            Link type: wireless
2017-01-02 02:25:43,727 - INFO #            inet6 addr: ff02::1/128  scope: local [multicast]
2017-01-02 02:25:43,733 - INFO #            inet6 addr: fe80::5855:4443:9918:b576/64  scope: local
2017-01-02 02:25:43,739 - INFO #            inet6 addr: ff02::1:ff18:b576/128  scope: local [multicast]
2017-01-02 02:25:43,744 - INFO #            inet6 addr: ff02::2/128  scope: local [multicast]
2017-01-02 02:25:43,745 - INFO #
```

smlng, 2017-01-10:
@kennethjiang from the tcpdump output above:

> 2017-01-02 02:25:42,318 - INFO # ndp: Router advertisement did not contain any source address information

I assume your RADVD config is not correct, please have a look at mine [here](https://github.com/RIOT-Makers/wpan-raspbian/wiki/Setup-native-6LoWPAN-router-using-Raspbian-and-RADVD#configure-6lr), the important option is `AdvSourceLLAddress on;`

smlng, 2017-01-10:
I often use a very similar setup and it works, however, I haven't tested with current master lately. So please report back if my config works for you ...

OlegHahm, 2017-01-10:
I remember that in earlier versions of radvd the L2 source address information was not included in the router advertisements.

kennethjiang, 2017-01-11:
You guys are probably right. I have `AdvSourceLLAddress on;` in `radvd.conf` but tcpdump does seem to show there is no source link address. @OlegHahm what version of radvd should I upgrade to?

kennethjiang, 2017-01-11:
Never mind I followed @smlng 's awesome tutorial at https://github.com/RIOT-Makers/wpan-raspbian/wiki/Setup-native-6LoWPAN-router-using-Raspbian-and-RADVD#configure-6lr and it worked!

closed 2017-01-11 by kennethjiang